e2nIEE/pandapower #2396: [bug] failure while executing test_from_mpc_m

quant12345, 2024-09-08:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
pytest pandapower/test/converter/test_from_mpc.py::test_from_mpc_m
```

### Issue Description and Traceback

I did [PR](https://github.com/e2nIEE/pandapower/pull/2395) and found in Actions that build (3.11, 1) fails.
I didn't see this when I ran tests locally. 
Tried running this test: `pytest pandapower/test/converter/test_from_mpc.py::test_from_mpc_m`
It is clear that the test is skipped, so I removed the test decorator and a message appeared that `matpowercaseframes` needs to be installed.

Then an error appeared (after installation matpowercaseframes):

```
pandapower/converter/pypower/from_ppc.py:135: 
AttributeError: 'bool' object has no attribute 'astype'
```
When printing the type `ppc['gen'][i, GEN_STATUS] > 0` there are `Python(bool)` values, and `astype` is a numpy attribute.
All values ​​were single, but since astype is used, and I assume that there can be several of them (array), otherwise you can get by with np.bool(value).
So I did it like this: `np.array(ppc['gen'][i, GEN_STATUS] > 0, dtype=np.bool)`

Then an error occurred: 
```
TypeError: loop of ufunc does not support argument 0 of type float which has no callable sqrt method
pandapower/converter/pypower/from_ppc.py:272:
```
When printing type `br_b**2 + br_g[is_trafo]**2`, on failure I got type `object`.

This error is easy to reproduce: `print(np.sqrt(np.array([1., 4.5, 7.0], dtype=object)))`

Having corrected this the tests passed successfully:
```
i0_percent = (np.sqrt(np.array(br_b ** 2 + br_g[is_trafo] ** 2,
                                      dtype=np.float64)) * 100 * baseMVA / sn)
```

### Expected Behavior

I want to make sure that I'm not the only one whose tests are failing. If so, I will fix it.

### Installed Versions

* python version: 3.12* pandas version:  2.2.2* networkx version: 3.3 * scipy version:  1.13.1* numpy version: 2.1.1 * packaging version:  24.1* tqdm version: 4.66.5 * deepdiff version:  7.0.1* Operating System name/version Ubuntu 24.04:

### Label

- [ ] Relevant labels are selected

yasirroni, 2024-09-10:
Hi @quant12345, numpy cant't have different dtypes across column by default, unlike pandas. In pandas, we can have a column as bool, another as range (used as index for example), and the other as float. Thus, if we write manually the .m file based on numpy, we normally write something like `1` instead of numpy `1.0`. Thus, I implement `.convert_dtypes()`.

What do you think?

quant12345, 2024-09-10:
@yasirroni 
I need time, I want to understand this better. Let's take a timeout for now, for a little while?

quant12345, 2024-09-10:
@yasirroni  @pawellytaev 
I updated the answer. Thoughts?

yasirroni, 2024-09-12:
@quant12345 after some though, some value such as `gen.PG` might be `Int64` due to how `.m` file parsed. On the other hand, it will most likely to be in `Float64` after run.

Or other example is `gen.QMAX`m where in `case9` it is `Int64` but in `case5` it is `Float64`.

___

My question though, what does pandapower requires?

1. The dtypes to be always consistent
2. dtypes need to be always float
3. dtypes can be mixed of int and float but using numpy dtypes, not the new pandas `Float64Dtype()` and `Int64Dtype()`

yasirroni, 2024-09-12:
Can you try pandapower compatibility with this function?

```python
def convert_dtypes_numpy_optimized(df):
    df = df.convert_dtypes()

    int_columns = df.select_dtypes(include=['integer']).columns
    df[int_columns] = df[int_columns].astype(np.int64)

    float_columns = df.select_dtypes(include=['float']).columns
    df[float_columns] = df[float_columns].astype(np.float64)

    string_columns = df.select_dtypes(include=['string']).columns
    df[string_columns] = df[string_columns].astype(str)

    bool_columns = df.select_dtypes(include=['boolean']).columns
    df[bool_columns] = df[bool_columns].astype(bool)

    return df
 ```
 
 I will make this optional, but I want to ensure that it also compatible with pandapower.

quant12345, 2024-09-12:
1. Question 1 implies that the types can be both pandas and numpy? If so, then when converting pandas types to numpy, all types will be object. Of course, this can be processed, but this will be additional overhead. And you need to remember that in numpy this will only be one type.
3. It will still be one type and there is no point in making them different (numpy array is created from several columns of dataframe with different types)

Based on the fact that the tests passed with matpowercaseframes1.0.8, it means everything was fine (the numpy array was float64). But, it is better to get an answer from the creators of pandapower, because if you still need both float and int types, then it will be numpy with the object type inside which are scalar float and int values ​​(it will take a long time to process) or stay on pandas.

To see all this visually, change the code in [_m2ppc](https://github.com/e2nIEE/pandapower/blob/11af12cd3db5a85135c7c022e8f5ac2e7a5b4086/pandapower/converter/matpower/from_mpc.py#L95)
<details>

<summary>_m2ppc(code)</summary>

```
def _m2ppc(mpc_file, casename_mpc_file):
    if not matpowercaseframes_imported:
        raise NotImplementedError(
            "matpowercaseframes is used to convert .m file. Please install that python "
            "package, e.g. via 'pip install matpowercaseframes'.")
    mpc_frames = CaseFrames(mpc_file)
    #'''
    for key in mpc_frames._attributes:
        if isinstance(mpc_frames.__getattribute__(key), pd.DataFrame):
            print(mpc_frames.__getattribute__(key).dtypes)# types of pandas
            print('type array numpy', mpc_frames.__getattribute__(key).values.dtype)
            #print(mpc_frames.__getattribute__(key))# show dataframe
            #print(mpc_frames.__getattribute__(key).values)# show array numpy
        else:
            print('scalar', mpc_frames.__getattribute__(key))
    #'''
    ppc = {key: mpc_frames.__getattribute__(key) if not isinstance(
        mpc_frames.__getattribute__(key), pd.DataFrame) else mpc_frames.__getattribute__(
        key).values for key in mpc_frames._attributes}
    _adjust_ppc_indices(ppc)
    return ppc
```

</details>

and run the test with printing: `pytest pandapower/test/converter/test_from_mpc.py::test_from_mpc_m -s`

And I also want to note if there is a dataframe with column types `float64 `and `int64`:
```
df = pd.DataFrame({'A':pd.Series([1.2, 3.0, 5.0, 7.0], dtype=np.float64),
                   'B': pd.Series([1, 55, 5, 10], dtype=np.int64)})

print(df.dtypes)
```
and when converted to numpy array, the resulting array is of type `np.float64`:

`print(df.values.dtype)# float64`

yasirroni, 2024-09-12:
@quant12345 Thanks for your reply.

1. I mean does pandapower accept that `gen.QMAX` sometimes `int` but on other time `float`?
3. I mean, does pandapower require matpowercaseframes return dataframe with all identical type, for example all `float` type? Does pandapower accept mix of `float` and `int`? In `1.0.8`, matpowercaseframes return DataFrame that does have mixed numpy types (float and int). Pandapower seems works this way, so that I can assume this to be True. Thus, using the `convert_dtypes_numpy_optimized` will not break pandapower.

___

If pandapower only deals with numpy using df.values, indeed it will be in single dtypes. But, if you interact column by column, for example `mpc.gen['QMAX']` or `df.['QMAX']`, it can return `int` while df.values return `float` dtype.

yasirroni, 2024-09-25:
Hi @quant12345, can you check to use the latest `v1.0.10`? `v1.0.9` is yanked in PyPI so that user should skip that version automatically.

This issue solved in https://github.com/UGM-EPSLab/matpowercaseframes/pull/11

quant12345, 2024-09-25:
Hi @yasirroni .  Checked: `pytest pandapower/test/converter/test_from_mpc.py::test_from_mpc_m`
The tests are passing successfully. 

I also looked at Actions fresh PRs that were launched today and there in 'List of installed packages': `matpowercaseframes 1.0.10.`

Should we close the issue?

yasirroni, 2024-09-26:
Yes, we should close this.

User that want consistency between read using `loadcase` and `parser` can use the `infer_numpy`. Others will use the old functionality. `v1.0.9` also yanked on PyPI.

closed 2024-09-26 by quant12345