e2nIEE/pandapower #2402: How to translate Reactive power in SVC definition

HafizRiaz, 2024-09-20:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Dear Authors, I have to use SVC for my time series simulation but the issue is that command to define SVC  "pandapower.create.create_svc(net, bus, x_l_ohm, x_cvar_ohm, set_vm_pu, thyristor_firing_angle_degree, name=None, controllable=True, in_service=True, index=None, min_angle_degree=90, max_angle_degree=180, **kwargs)" does not contain Qsvc directly and formula given in the docs contains complex voltage , i am wondering how to translate a real reactive power in terms of complex voltage ? e.g. i have to install 1KVAR svc, hoe can i define it in above parameters>?

### Label

- [X] Relevant labels are selected

HafizRiaz, 2024-11-25:
> Hey @HafizRiaz<https://github.com/HafizRiaz>, the SVC is basically a shunt element. Thus the Q is dependent on the voltage, which is solved during the Newton Raphson iterations during the power flow.
>
> So by setting the x_l_ohm, thyristor_firing_angle_degree, etc., you implicitly set the rated Q for the SVC. If you want to have the Q as a direct input, you can use the shunt and use a shunt controller for setting the voltage set point.

Ok got it. Thanks

pawellytaev, 2024-11-26:
Hey, closing this issue then 👍

closed 2024-11-26 by pawellytaev