e2nIEE/pandapower #2403: [bug] The identified bug of function pandapower.converter.from_mpc for .m file

Len02805, 2024-09-23:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
net = pp.converter.from_mpc('/Your/path/to/Matpower/Cases/case69.m', f_hz=50, validate_conversion=False)
pp.runpp(net)
```

### Issue Description and Traceback

The original file case69.m from https://github.com/MATPOWER/matpower/blob/master/data/case69.m cannot be converted to the right pp network.  Actually, a lof of .m file under this dir (github.com/MATPOWER/matpower/blob/master/data) cannot be successfully processed. 

This is becase the following code para in the ending of .m file is not corrected processed by "from_mpc" function:

```

%% convert brach impedances from Ohms to p.u.
[PQ, PV, REF, NONE, BUS_I, BUS_TYPE, PD, QD, GS, BS, BUS_AREA, VM, ...
    VA, BASE_KV, ZONE, VMAX, VMIN, LAM_P, LAM_Q, MU_VMAX, MU_VMIN] = idx_bus;
[F_BUS, T_BUS, BR_R, BR_X, BR_B, RATE_A, RATE_B, RATE_C, ...
    TAP, SHIFT, BR_STATUS, PF, QF, PT, QT, MU_SF, MU_ST, ...
    ANGMIN, ANGMAX, MU_ANGMIN, MU_ANGMAX] = idx_brch;
Vbase = mpc.bus(1, BASE_KV) * 1e3;      %% in Volts
Sbase = mpc.baseMVA * 1e6;              %% in VA
mpc.branch(:, [BR_R BR_X]) = mpc.branch(:, [BR_R BR_X]) / (Vbase^2 / Sbase);

%% convert loads from kW to MW
mpc.bus(:, [PD, QD]) = mpc.bus(:, [PD, QD]) / 1e3;

```

So for this .m file, if you add the follwoing code to /opt/anaconda3/envs/env_name/lib/python3.9/site-packages/pandapower/converter/matpower/from_mpc.py after the code line 100 

```
    if mpc_file[-8:]=="case69.m":
        print("Activated for case 69")
        baseMVA69 = 10
        Sbase69 = baseMVA69 * 1e6
        Vbase69 = 12.66 * 1e3
        mpc_frames.branch[["BR_R", "BR_X"]] = mpc_frames.branch[["BR_R", "BR_X"]] / (Vbase69**2 / Sbase69)
        mpc_frames.bus[["PD", "QD"]] = mpc_frames.bus[["PD", "QD"]]/1000
```

Then the reproducible Example can be successfully run by pandapower as smoothly and accurately as the matlab :)

However, this is the customized method for this spcific .m file. It would be helpful if you guys could rewrite a more general method.

### Expected Behavior

Fix the bug as I mentioned :)

### Installed Versions

INSTALLED VERSIONS
------------------
commit                : d9cdd2ee5a58015ef6f4d15c7226110c9aab8140
python                : 3.9.15.final.0
python-bits           : 64
OS                    : Darwin
OS-release            : 23.3.0
Version               : Darwin Kernel Version 23.3.0: Thu Dec 21 02:29:41 PST 2023; root:xnu-10002.81.5~11/RELEASE_ARM64_T8122
machine               : arm64
processor             : arm
byteorder             : little
LC_ALL                : None
LANG                  : None
LOCALE                : None.UTF-8

pandas                : 2.2.2
numpy                 : 1.23.5
pytz                  : 2024.1
dateutil              : 2.9.0.post0
setuptools            : 73.0.0
pip                   : 24.0
Cython                : None
pytest                : 8.3.2
hypothesis            : None
sphinx                : 7.4.7
blosc                 : None
feather               : None
xlsxwriter            : 3.2.0
lxml.etree            : None
html5lib              : None
pymysql               : None
psycopg2              : 2.9.9
jinja2                : 3.1.4
IPython               : 8.18.1
pandas_datareader     : None
adbc-driver-postgresql: None
adbc-driver-sqlite    : None
bs4                   : 4.12.3
bottleneck            : 1.3.7
dataframe-api-compat  : None
fastparquet           : 2024.5.0
fsspec                : 2024.6.1
gcsfs                 : None
matplotlib            : 3.8.4
numba                 : 0.60.0
numexpr               : 2.8.7
odfpy                 : None
openpyxl              : 3.1.5
pandas_gbq            : None
pyarrow               : 17.0.0
pyreadstat            : None
python-calamine       : None
pyxlsb                : None
s3fs                  : None
scipy                 : 1.13.1
sqlalchemy            : None
tables                : None
tabulate              : 0.9.0
xarray                : 2024.7.0
xlrd                  : 2.0.1
zstandard             : None
tzdata                : 2023.3
qtpy                  : 2.4.1
pyqt5                 : None

### Label

- [X] Relevant labels are selected

yasirroni, 2024-09-24:
Hi, I'm matpowercaseframes and matpower-pip developer. The easiest way to solve it is to run matpower-pip.

```python
from matpower import start_instance

from matpowercaseframes import CaseFrames

m = start_instance()
CASE_NAME = "case16am.m"
cf_16am = CaseFrames(CASE_NAME)
```

Then, after the line 100, we should introduce something like

```python
ppc = {key: mpc_frames.__getattribute__(key) if not isinstance(
        mpc_frames.__getattribute__(key), pd.DataFrame) else mpc_frames.__getattribute__(
        key).values for key in mpc_frames._attributes}
    _adjust_ppc_indices(ppc)
```

See: https://github.com/UGM-EPSLab/matpowercaseframes/blob/master/notebooks/load_case16am.ipynb

Let's wait pandapower developer on the best way to integrate this to pandapower.

Len02805, 2024-09-24:
Hi Yasirroni,

I know the m = start_instance() should work as you mentioned in https://github.com/e2nIEE/pandapower/issues/1556. 

However, as you also mentioned in the above link "Sadly, it need a third parties octave.cli." and actually the [oct2py](https://pypi.org/project/oct2py/) package is unstable as well. BTW, I had a bug with function start_instance(). But I do not have enough time and interest to investigate the reason as it doesn't affect my work.

`error: graphics_toolkit: = toolkit is not available
error: called from
    graphics_toolkit at line 84 column 5`

Sadly, we fall in a loop between two fairly unstable dependency packages while using pandapower :(

yasirroni, 2024-09-24:
Handling per name file like  `if mpc_file[-8:]=="case69.m":` is not wise and I think it should not be implemented.

In my case, `oct2py` is pretty stable and reliable. If you got a problem in installing matpower-pip or oct2py, you should report the issue in their respective repo. You can mention me there with complete debugging or info if possible.

Len02805, 2024-09-24:
Yes, as I mentioned, "this is a customized method for this spcific .m file. It would be helpful if you guys could rewrite a more general method". I post this spcific method for insight the debug solution. Appreantly you cannot implement this into the pandapower public package.

yasirroni, 2024-09-25:
`pandapower` relies on `matpowercaseframes` to read `.m` file. Solution to read that file is already solved with `matpowercaseframes` and `matpower-pip`.

If you don't want to use `matpowercaseframes` and `matpower-pip`, run `matlab`/`octave`, iterate loop save as `.mat`, then `pandapower` will be able to use `scipy.io`.

I think that is the best possible way here.

Len02805, 2024-09-25:
If you run the reproducible example above, you will get an error.

yasirroni, 2024-09-25:
> If you run the reproducible example above, you will get an error.

Which code? Which error?

yasirroni, 2024-09-25:
```python
from matpower import start_instance

from matpowercaseframes import CaseFrames

m = start_instance()
CASE_NAME = "case16am.m"
cf_16am = CaseFrames(CASE_NAME)
```
```python
import numpy as np
import pandas as pd

def _adjust_ppc_indices(ppc):
    # adjust indices of ppc, since ppc must start at 0 rather than 1 (matlab)
    ppc["bus"][:, 0] -= 1
    ppc["branch"][:, 0] -= 1
    ppc["branch"][:, 1] -= 1
    # if in ppc is only one gen -> numpy initially uses one dim array -> change to two dim array
    if len(ppc["gen"].shape) == 1:
        ppc["gen"] = np.array(ppc["gen"], ndmin=2)
    ppc["gen"][:, 0] -= 1

mpc_frames = cf_16am
ppc = {key: mpc_frames.__getattribute__(key) if not isinstance(
        mpc_frames.__getattribute__(key), pd.DataFrame) else mpc_frames.__getattribute__(
        key).values for key in mpc_frames._attributes}
_adjust_ppc_indices(ppc)
print(ppc)
```
```text
{'version': '2',
 'baseMVA': 10,
 'bus': array([[ 0.000e+00,  3.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.000e+00,  1.000e+00],
        [ 1.000e+00,  1.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 2.000e+00,  1.000e+00,  2.000e+03,  1.600e+03,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 3.000e+00,  1.000e+00,  3.000e+03,  4.000e+02,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 4.000e+00,  1.000e+00,  2.000e+03, -4.000e+02,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 5.000e+00,  1.000e+00,  1.500e+03,  1.200e+03,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 6.000e+00,  1.000e+00,  4.000e+03,  2.700e+03,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 7.000e+00,  1.000e+00,  5.000e+03,  1.800e+03,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 8.000e+00,  1.000e+00,  1.000e+03,  9.000e+02,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 9.000e+00,  1.000e+00,  6.000e+02, -5.000e+02,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 1.000e+01,  1.000e+00,  4.500e+03, -1.700e+03,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 1.100e+01,  1.000e+00,  1.000e+03,  9.000e+02,  0.000e+00,
          0.000e+00,  1.000e+00,  1.000e+00,  0.000e+00,  1.266e+01,
          1.000e+00,  1.100e+00,  9.000e-01],
        [ 1.200e+01,  1.000e+00,  1.000e+03, -1.100e+03,  0.000e+00,
[... 29 lines cut ...]
          0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,
          1.000e+00, -3.600e+02,  3.600e+02],
        [ 6.000e+00,  8.000e+00,  1.763e-01,  1.763e-01,  0.000e+00,
          0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,
          1.000e+00, -3.600e+02,  3.600e+02],
        [ 7.000e+00,  9.000e+00,  1.763e-01,  1.763e-01,  0.000e+00,
          0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,
          1.000e+00, -3.600e+02,  3.600e+02],
        [ 7.000e+00,  1.000e+01,  1.282e-01,  1.763e-01,  0.000e+00,
          0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,
          1.000e+00, -3.600e+02,  3.600e+02],
        [ 1.000e+00,  1.100e+01,  1.763e-01,  1.763e-01,  0.000e+00,
          0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,
          1.000e+00, -3.600e+02,  3.600e+02],
        [ 1.100e+01,  1.200e+01,  1.442e-01,  1.923e-01,  0.000e+00,
          0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,
          1.000e+00, -3.600e+02,  3.600e+02],
        [ 1.100e+01,  1.300e+01,  1.282e-01,  1.763e-01,  0.000e+00,
          0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,
          1.000e+00, -3.600e+02,  3.600e+02],
        [ 1.300e+01,  1.400e+01,  6.410e-02,  6.410e-02,  0.000e+00,
          0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,  0.000e+00,
          1.000e+00, -3.600e+02,  3.600e+02]]),
 'gencost': array([[ 2,  0,  0,  3,  0, 20,  0]])}
```

Len02805, 2024-09-25:
```
import pandapower as pp
net = pp.converter.from_mpc('/Your/path/to/Matpower/Cases/case69.m', f_hz=50, validate_conversion=False)
pp.runpp(net)

```

```
FutureWarning: The behavior of DataFrame concatenation with empty or all-NA entries is deprecated. In a future version, this will no longer exclude empty or all-NA columns when determining the result dtypes. To retain the old behavior, exclude the relevant entries before the concat operation.
  net[table] = pd.concat([net[table], dd[dd.columns[~dd.isnull().all()]]], sort=False)
/opt/anaconda3/envs/Envs/lib/python3.9/site-packages/pandapower/converter/pypower/from_ppc.py:277: FutureWarning: Setting an item of incompatible dtype is deprecated and will raise an error in a future version of pandas. Value '[]' has dtype incompatible with int64, please explicitly cast to a compatible dtype first.
  branch_lookup.loc[~is_line, "element"] = idx_trafo
---------------------------------------------------------------------------
LoadflowNotConverged                      Traceback (most recent call last)
Cell In[2], line 3
      1 network = pp.converter.from_mpc('case69.m', f_hz=50, validate_conversion=False)
      2 network.line["max_i_ka"]=0.1
----> 3 pp.runpp(network)

File /opt/anaconda3/envs/Envs/lib/python3.9/site-packages/pandapower/run.py:250, in runpp(net, algorithm, calculate_voltage_angles, init, max_iteration, tolerance_mva, trafo_model, trafo_loading, enforce_q_lims, check_connectivity, voltage_depend_loads, consider_line_temperature, run_control, distributed_slack, tdpf, tdpf_delay_s, **kwargs)
    248 _check_bus_index_and_print_warning_if_high(net)
    249 _check_gen_index_and_print_warning_if_high(net)
--> 250 _powerflow(net, **kwargs)

File /opt/anaconda3/envs/Envs/lib/python3.9/site-packages/pandapower/powerflow.py:86, in _powerflow(net, **kwargs)
     84 result = _run_pf_algorithm(ppci, net["_options"], **kwargs)
     85 # read the results (=ppci with results) to net
---> 86 _ppci_to_net(result, net)

File /opt/anaconda3/envs/Envs/lib/python3.9/site-packages/pandapower/powerflow.py:190, in _ppci_to_net(result, net)
    188     algorithm = net["_options"]["algorithm"]
    189     max_iteration = net["_options"]["max_iteration"]
--> 190     raise LoadflowNotConverged("Power Flow {0} did not converge after "
    191                                "{1} iterations!".format(algorithm, max_iteration))
    192 else:
    193     net["_ppc"] = result

LoadflowNotConverged: Power Flow nr did not converge after 10 iterations!
```

matpowercaseframes==1.0.8
matpower==8.0.0.2.1.8
pandapower==2.14.11

yasirroni, 2024-09-25:
```python
from matpower import start_instance

from matpowercaseframes import CaseFrames

m = start_instance()
CASE_NAME = "case16am.m"
cf = CaseFrames(CASE_NAME, load_case_engine=m)

import numpy as np
import pandas as pd

def _adjust_ppc_indices(ppc):
    # adjust indices of ppc, since ppc must start at 0 rather than 1 (matlab)
    ppc["bus"][:, 0] -= 1
    ppc["branch"][:, 0] -= 1
    ppc["branch"][:, 1] -= 1

    # if in ppc is only one gen -> numpy initially uses one dim array -> change to two dim array
    if len(ppc["gen"].shape) == 1:
        ppc["gen"] = np.array(ppc["gen"], ndmin=2)
    ppc["gen"][:, 0] -= 1

mpc_frames = cf
ppc = {key: mpc_frames.__getattribute__(key) if not isinstance(
        mpc_frames.__getattribute__(key), pd.DataFrame) else mpc_frames.__getattribute__(
        key).values for key in mpc_frames._attributes}
_adjust_ppc_indices(ppc)

import pandapower as pp
import pandapower.converter as pc

net = pc.from_ppc(ppc)
pp.runpp(net)
```

Len02805, 2024-09-25:
Yes. This is supposed to be that the following codes in the ending of .m file is not corrected processed by "from_mpc" function:

For your "case16am.m" case, the para should be:

```
%% convert branch impedances from Ohms to p.u.
[PQ, PV, REF, NONE, BUS_I, BUS_TYPE, PD, QD, GS, BS, BUS_AREA, VM, ...
    VA, BASE_KV, ZONE, VMAX, VMIN, LAM_P, LAM_Q, MU_VMAX, MU_VMIN] = idx_bus;
[F_BUS, T_BUS, BR_R, BR_X, BR_B, RATE_A, RATE_B, RATE_C, ...
    TAP, SHIFT, BR_STATUS, PF, QF, PT, QT, MU_SF, MU_ST, ...
    ANGMIN, ANGMAX, MU_ANGMIN, MU_ANGMAX] = idx_brch;
Vbase = mpc.bus(1, BASE_KV) * 1e3;      %% in Volts
Sbase = mpc.baseMVA * 1e6;              %% in VA
mpc.branch(:, [BR_R BR_X]) = mpc.branch(:, [BR_R BR_X]) / (Vbase^2 / Sbase);

%% convert loads from kW to MW
mpc.bus(:, [PD, QD]) = mpc.bus(:, [PD, QD]) / 1e3;

```

> Which code? Which error?

Now you find and reproduce the code and error :)

yasirroni, 2024-09-26:
> Now you find and reproduce the code and error :)

@Len02805 No, it seems you don't understand it yet. `load_case_engine` already execute the MATLAB code of:

```matlab
%% convert branch impedances from Ohms to p.u.
[PQ, PV, REF, NONE, BUS_I, BUS_TYPE, PD, QD, GS, BS, BUS_AREA, VM, ...
    VA, BASE_KV, ZONE, VMAX, VMIN, LAM_P, LAM_Q, MU_VMAX, MU_VMIN] = idx_bus;
[F_BUS, T_BUS, BR_R, BR_X, BR_B, RATE_A, RATE_B, RATE_C, ...
    TAP, SHIFT, BR_STATUS, PF, QF, PT, QT, MU_SF, MU_ST, ...
    ANGMIN, ANGMAX, MU_ANGMIN, MU_ANGMAX] = idx_brch;
Vbase = mpc.bus(1, BASE_KV) * 1e3;      %% in Volts
Sbase = mpc.baseMVA * 1e6;              %% in VA
mpc.branch(:, [BR_R BR_X]) = mpc.branch(:, [BR_R BR_X]) / (Vbase^2 / Sbase);

%% convert loads from kW to MW
mpc.bus(:, [PD, QD]) = mpc.bus(:, [PD, QD]) / 1e3;
```

The problem is not that. See this:

```
from matpower import start_instance

from matpowercaseframes import CaseFrames

m = start_instance()
CASE_NAME = "case16am.m"

print(
    CaseFrames(CASE_NAME).branch[["BR_R", "BR_X"]],
    '\n',
    CaseFrames(CASE_NAME, load_case_engine=m).branch[["BR_R", "BR_X"]],
)
```

```text
      BR_R          BR_X
1   0.0000  1.000000e-08
2   0.1202  1.603000e-01
3   0.1282  1.763000e-01
4   0.1442  2.885000e-01
5   0.0641  6.410000e-02
6   0.1763  1.763000e-01
7   0.1282  1.763000e-01
8   0.1763  1.763000e-01
9   0.1763  1.763000e-01
10  0.1282  1.763000e-01
11  0.1763  1.763000e-01
12  0.1442  1.923000e-01
13  0.1282  1.763000e-01
14  0.0641  6.410000e-02 
         BR_R          BR_X
1   0.000000  6.239253e-10
2   0.007500  1.000152e-02
3   0.007999  1.099980e-02
4   0.008997  1.800024e-02
5   0.003999  3.999361e-03
6   0.011000  1.099980e-02
7   0.007999  1.099980e-02
8   0.011000  1.099980e-02
9   0.011000  1.099980e-02
10  0.007999  1.099980e-02
11  0.011000  1.099980e-02
12  0.008997  1.199808e-02
13  0.007999  1.099980e-02
14  0.003999  3.999361e-03
```

I also run your "workaround",

```python
    if mpc_file[-8:]=="case69.m":
        print("Activated for case 69")
        baseMVA69 = 10
        Sbase69 = baseMVA69 * 1e6
        Vbase69 = 12.66 * 1e3
        mpc_frames.branch[["BR_R", "BR_X"]] = mpc_frames.branch[["BR_R", "BR_X"]] / (Vbase69**2 / Sbase69)
        mpc_frames.bus[["PD", "QD"]] = mpc_frames.bus[["PD", "QD"]]/1000
```

But, it still resulted in same error.

```python
from matpower import start_instance

from matpowercaseframes import CaseFrames

m = start_instance()
CASE_NAME = "case16am.m"
cf = CaseFrames(CASE_NAME)

mpc_frames = cf

baseMVA69 = 10
Sbase69 = baseMVA69 * 1e6
Vbase69 = 12.66 * 1e3
mpc_frames.branch[["BR_R", "BR_X"]] = mpc_frames.branch[["BR_R", "BR_X"]] / (Vbase69**2 / Sbase69)
mpc_frames.bus[["PD", "QD"]] = mpc_frames.bus[["PD", "QD"]]/1000

import numpy as np
import pandas as pd

def _adjust_ppc_indices(ppc):
    # adjust indices of ppc, since ppc must start at 0 rather than 1 (matlab)
    ppc["bus"][:, 0] -= 1
    ppc["branch"][:, 0] -= 1
    ppc["branch"][:, 1] -= 1

    # if in ppc is only one gen -> numpy initially uses one dim array -> change to two dim array
    if len(ppc["gen"].shape) == 1:
        ppc["gen"] = np.array(ppc["gen"], ndmin=2)
    ppc["gen"][:, 0] -= 1

ppc = {key: mpc_frames.__getattribute__(key) if not isinstance(
        mpc_frames.__getattribute__(key), pd.DataFrame) else mpc_frames.__getattribute__(
        key).values for key in mpc_frames._attributes}
_adjust_ppc_indices(ppc)

import pandapower as pp
import pandapower.converter as pc

net = pc.from_ppc(ppc, f_hz=60, validate_conversion=False)
if "mpc_additional_data" in ppc:
    if "_options" not in net:
        net["_options"] = dict()
    net._options.update(ppc["mpc_additional_data"])

try:
    pp.runpp(net)
except Exception as e:
    print(e)
```

yasirroni, 2024-09-26:
After checking, even MATPOWER itself did not converge. Using other file, that is `case69.m`, my example is working.

```python
from matpower import start_instance

from matpowercaseframes import CaseFrames

m = start_instance()
CASE_NAME = 'case69.m`
cf = CaseFrames(CASE_NAME, load_case_engine=m)

mpc_frames = cf

if CASE_NAME == 'case16am.m':
    baseMVA69 = 10
    Sbase69 = baseMVA69 * 1e6
    Vbase69 = 12.66 * 1e3
    mpc_frames.branch[["BR_R", "BR_X"]] = mpc_frames.branch[["BR_R", "BR_X"]] / (Vbase69**2 / Sbase69)
    mpc_frames.bus[["PD", "QD"]] = mpc_frames.bus[["PD", "QD"]]/1000

import numpy as np
import pandas as pd

def _adjust_ppc_indices(ppc):
    # adjust indices of ppc, since ppc must start at 0 rather than 1 (matlab)
    ppc["bus"][:, 0] -= 1
    ppc["branch"][:, 0] -= 1
    ppc["branch"][:, 1] -= 1

    # if in ppc is only one gen -> numpy initially uses one dim array -> change to two dim array
    if len(ppc["gen"].shape) == 1:
        ppc["gen"] = np.array(ppc["gen"], ndmin=2)
    ppc["gen"][:, 0] -= 1

ppc = {key: mpc_frames.__getattribute__(key) if not isinstance(
        mpc_frames.__getattribute__(key), pd.DataFrame) else mpc_frames.__getattribute__(
        key).values for key in mpc_frames._attributes}
_adjust_ppc_indices(ppc)

import pandapower as pp
import pandapower.converter as pc

net = pc.from_ppc(ppc, f_hz=60, validate_conversion=False)
if "mpc_additional_data" in ppc:
    if "_options" not in net:
        net["_options"] = dict()
    net._options.update(ppc["mpc_additional_data"])

try:
    pp.runpp(net)
except Exception as e:
    print(e)
```

If `cf = CaseFrames(CASE_NAME, load_case_engine=m)` changed to `cf = CaseFrames(CASE_NAME)` (currently used by `pandapower`), the NR will not converge.

So, the correct use of `matpowercaseframes` in this case is to use `load_case_engine`.

yasirroni, 2024-09-26:
As mentioned in https://github.com/MATPOWER/matpower/issues/208, `case16am.m` is non convergent case. My bad changing your initial `case69.m` to `case16am.m`.

Len02805, 2024-09-29:
Yes, it seems you understand this bug in Pandapower on using matpowercaseframes (load_case_engine).

yasirroni, 2025-05-10:
```python
from matpower import path_matpower_cases, start_instance
import pandapower as pp

m = start_instance()
net = pp.converter.from_mpc(f'{path_matpower_cases}/case69.m', f_hz=50, validate_conversion=False, load_case_engine=m)
pp.runpp(net)
```

yasirroni, 2025-05-10:
Hi @vogt31337  and @Len02805, can you review and test the PR? It should fix this issue. Me personally, wanted panda power to be able to pass all arguments in https://github.com/yasirroni/matpowercaseframes/blob/8652342f894000ee42453409971ce8035f72cb19/matpowercaseframes/core.py#L22-L32. But, I think, it would be better for me in `matpowercaseframes` to support export to `pp` and `ppc` format.

yasirroni, 2025-05-27:
Hi @vogt31337. Yes, the PR in https://github.com/e2nIEE/pandapower/pull/2606 will support executing arbitrary matlab/octave code.

As far as I know, there is no parser able to run matlab/octave code without actually running it.
_
Using matlab/octave engine is optional, just for a case that require running matlab/octave code and when the user confident to use it (default to None, not using any engine.).

vogt31337, 2025-05-28:
@yasirroni, we discussed this issue. I think the main problem we are having with this, is the arbitrary code execution. 
So your proposed solution would be to incorporate a matlab / octave engine which after loading a .m file executes the code. From a user perspective, you will just load the file and if it does not load, you will start the matlab / octave engine in the hopes, that the file then loads.
So basically this opens the door to malicious code, to be loaded... May be I am a bit over hesitant here, but it reminds me a lot of for example this [Untitled Goose Game - Insecure Deserialization](https://pulsesecurity.co.nz/advisories/untitled-goose-game-deserialization).

Chat GPT 4 also considers this a bad practice:
```
Storing executable code in a data file and then loading and executing it is generally considered a bad practice due to several reasons:

1. Security Risks: As you've mentioned, this can lead to the execution of malicious code if the data file is manipulated.
2. Maintainability: It can make the system harder to understand and maintain. Code should be in source files, and data should be in data files.
3. Portability: It can restrict the portability of your data. If the code is specific to a certain platform or language, the data will not be portable to a different platform or language.

It's recommended to separate code and data.
```

Therefore I would urge you as a developer of matpowercaseframes and matpower-pip to consider removing the need to execute code from your save format.

vogt31337, 2025-05-28:
@yasirroni and @Len02805 firstly thanks for the discussion so far!

@yasirroni, I understand your point. I think matlab/octave is not safe incoherently, I just think code and data should be seperated in a save format. Therefore probably rethink the way these cases are stored.

Thank you for revisiting your work and trying another solution. I really appreciate the effort you put in! 
Probably a ipnyb could help, in showing how you would perform the loading. And maybe we put a disclaimer in the matpower converter, that it will not execute code automatically and then link to the ipnyb notebook (which contains all the ways you mentioned already and maybe a disclaimer).