e2nIEE/pandapower #2406: [feature] Request to add four-winding transformer equipment

zhtee, 2024-09-25:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [X] Adding new functionality to pandapower

- [ ] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

Hello, in our practical work, four-winding transformers have been increasingly used. This type of transformer adds a filtering winding to achieve functions such as filtering third harmonics. Therefore, I would like to know if it is possible to add a four-winding transformer device?

### Feature Description

Adding four-winding transformer equipment eliminates the need for users to perform equivalent calculations themselves.

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-11-21:
@zhtee So far I have not heard of such transformer, could you provide literature explaining the details? Implementation of 3 winding transformers was quite complicated and is based on a reduction to three 2w transfomer. Something similar will be needed. And pandapower does not compute harmonics, so what would the benefit be?

vogt31337, 2024-11-26:
@zhtee seems to be quite complicated to implement, sadly I do not need it right now. You are welcomed to create a Pull Request.

zhtee, 2024-11-26:
@vogt31337 I fully understand and hope to contribute to PandaPower in the future. Thank you again for your reply!

closed 2024-11-26 by zhtee