e2nIEE/pandapower #2407: [feature] Distance protection

ricoeo, 2024-09-25:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [X] Adding new functionality to pandapower

- [ ] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

I have read many papers regarding the protection in pandapower. In these papers, it is stated that distance protection has been implemented. But I don't see where this statement comes from. I expect to see this protection device type in the new release.

### Feature Description

Implementation and tutorials about distance protection type

### Label

- [ ] Relevant labels are selected

ricoeo, 2024-09-27:
> Hey @ricoeo ,
> 
> The distance relay has already been implemented and we are in the testing phase now to make it an open-source. We will release it as soon as we fulfill the requirement to make it public.
> 
> I do appreciate your interest and hope that you will use pandapower distance protection in the near future.
> 
> Regards, Gourab Banerjee

Thanks for your reply. I am right now implementing distance protection for my thesis work. May I ask some questions here?
In case it is allowed. 
[image]
A simple example is drawn. The bus has index of 0,1,2 from left to right. The distance protection device is located at bus 0 towards the direction of bus 1. Line 1 has the parameters: from_bus 0 and to_bus 1, line_id 0.
**A three-phase fault is simulated in the middle bus.**
`sc.calc_sc(net, fault="3ph", bus =1, branch_results=True, return_all_currents=True)`
then we will get the result of `net.res_bus_sc` and `net.res_line_sc`
The following part gives my understanding of the calculation of the distance. And I don't get the result I expect.
`vm_pu = net.res_line_sc.loc[0]["vm_from_pu"].item()`
`ikss_ka = net.res_line_sc.loc[0]["ikss_from_ka"].item()`
 `va_degree = net.res_line_sc.loc[0]["va_from_degree"].item()`
`ikss_degree = net.res_line_sc.loc[0]["ikss_from_degree"].item()`
From the measurement, the impedance is calculated:
`mag_sensed = vm_pu * HV * 1e-3 / ikss_ka`
 `angle_sensed = va_degree - ikss_degree`
May you point out what the problem is of implementing like this?

pawellytaev, 2024-10-14:
Hi @ricoeo, thanks for your interest in the implementations!
With your approach I don't see a problem. You need to be careful with the consideration of the voltage (which is should be rated voltage divided by sqrt(3), otherwise you calculate the impedance with the line-to-line voltage).

In your example the distance, i.e., the impedance between the short-circuit and the distance relay should be equal to the line impedance. Therefore you can easily verify your approach by using the line R & X parameters. 
```
r = net.line.loc[1, 'r_ohm_per_km'] * net.line.loc[1, 'length_km']
x = net.line.loc[1, 'x_ohm_per_km'] * net.line.loc[1, 'length_km']
z = r + 1j*x
```

`mag_sensed` should then be equal to `np.abs(z)`, `angle_sensed` should then be equal to `np.angle(z)`.

closed 2024-10-15 by ricoeo