e2nIEE/pandapower #2408: [BUG] powerflow leads to error

AnkurArohi, 2024-09-25:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
>>> pandapower.rundcpp(topo)
 RuntimeWarning: invalid value encountered in divide
  V0[gbus] = gen[on, VG] / abs(V0[gbus]) * V0[gbus]

I tried only DC run pp and I have this runtime warning, all the attributes seems ok, DC powerflow should provide better warnings in irder to find where the error is
```

### Issue Description and Traceback

The slack is already set and the balance looks good, atleast DC load flow should work.

### Expected Behavior

DC loadflow should work or the exception should be more descriptive 

### Installed Versions

* python version: * pandas version:  * networkx version:  * scipy version:  * numpy version:  * packaging version:  * tqdm version:  * deepdiff version:  * Operating System name/version:
python 3,11
pandapower 2.14.6
pandas 2.2.1

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-11-21:
@AnkurArohi, have you tried the diagnostic function?

AnkurArohi, 2024-11-22:
yes and it is all good

vogt31337, 2026-11-11:
Closed due to inactivity.

closed 2026-11-11 by vogt31337