e2nIEE/pandapower #2409: [bug] YOUR TITLE

SimonUlm, 2024-09-27:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
from pandapower.converter.matpower.from_mpc import _m2ppc

ppc_before = _m2ppc('pglib_opf_case24_ieee_rts.m', 'mpc')
net = pp.converter.from_mpc(file)
ppc_after = pp.converter.to_ppc(net, init='flat')
```

### Issue Description and Traceback

### Note ###
The file mentioned above is from the [pglib-opf](https://github.com/power-grid-lib/pglib-opf) library.

### Issue ###
When comparing `ppc_before['branch']` and `ppc_after['branch']`, you will notice that the _from_ and _to_ buses of all transformers have been swapped. However, the tap ratios remain the same, which will result in **wrong results in pf and opf calculations**.
Afaik this is due to _from_ppc.py_ (lines 241 to 245)  where `hv_bus` and `lv_bus` are swapped in case the base voltage of the _from_ bus is less than the base voltage if the _to_ bus, but the tap ratio and phase shift remain the same.

### Expected Behavior

The pandapower case structure should contain the correct direction of the transformer or at least contain information to restore the original direction.

### Installed Versions

* python version: * pandas version:  * networkx version:  * scipy version:  * numpy version:  * packaging version:  * tqdm version:  * deepdiff version:  * Operating System name/version:

### Label

- [X] Relevant labels are selected

vogt31337, 2024-10-21:
Hi, interesting error. I'll discuss it with some people. I am wondering that this problem hasn't surfaced earlier, since pandapower is basically built around the matpower architecture. And I am confused why the tests don't capture it...

vogt31337, 2024-10-21:
Discussed this a little, seems to be a really weird setup, since normally you would connect the HV-Side of a transformer to the higher voltage in a grid model and the LV-Side of the transformer to the lower voltage. Your suggest that this was flipped in your example and therefore errors occured. I'd suggest repairing your examples / test cases. But still this should be at least communicated to the user.

SimonUlm, 2024-11-04:
Hi, thank you very much for your reply and sorry for my very late response. I agree that this setting is rather unusual. However, this wasn't just any test case, but a pglib-opf (former NESTA) test case which are considered state-of-the-art benachmarking test cases by many. (Especially when it comes to convex relaxation, which is admittedly not the focus of pandapower, yet). Also, this wasn't the only test case having this issue. Therefore, I think this issue should be addressed. Either by communicating it to the user (as you already suggested) or recalculate the tap ratio. Personally, I would prefer the latter.

vogt31337, 2024-11-21:
@SimonUlm, no problem. I still find it kind of weird. As this issue has only now arisen, with a test suite. I would rather question the vadility of this weird setups in these tests. But I'll take it into consideration for fixing.

vogt31337, 2024-12-09:
@SimonUlm I am currently digging into this topic, thanks for the hints. The matrices seem to differ a lot more than I thought. Not only are overratings are set (seems to be some basic assumptions pandapower is doing), but for example branch 2 only exists once in the after matrix, but 2 times in the before matrix.
I'll keep investigating, but seems to be a complex one...

vogt31337, 2024-12-09:
Update, the missing branch seems to be a trafo from bus 2 to bus 23 (or vice versa), which is quite shocking, that it's missing...

SimonUlm, 2025-01-09:
Hi, adding the test for validtating whether the transformers have been swapped is probably a good idea, thanks. As for me personally this is sufficient, I will close this issue. Thank you.

closed 2025-01-09 by SimonUlm