e2nIEE/pandapower #2413: [feature] KeyError: 'bus_dc'

quant12345, 2024-10-06:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [ ] Adding new functionality to pandapower

- [X] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

In the julia build (3.9, 2) I see three tests with an error: `KeyError: 'bus_dc'`.
This happens because a non-existent key is being accessed.
`results.py -> _get_aranged_lookup - row 83`

this line:
`if len(net[bus_table]) == 0:`

Running the test: ` pytest pandapower/test/opf/test_pandamodels_runpm.py::test_runpm_multi_vstab`

### Feature Description

suggesting to replace with:

```
if bus_table not in net:
    net.setdefault(bus_table)
```

But the code still ends with a different error. It is not clear how the data for the bus_dc table is generated?

### Label

- [ ] Relevant labels are selected

yasirroni, 2024-10-09:
Hi @quant12345, can you test this on `matpowercaseframes` to? PowerModels accept arbitrary data as far as I know and `matpowercaseframes` not implemented that yet.

quant12345, 2024-10-09:
Hi @yasirroni, my modest knowledge in this area is not enough. I have not yet figured out how the data is formed. I can assume that a dataframe should be created in the 'bus_dc' table, but it is not clear where it is formed. The error clearly says that there is no such key in the dictionary.
You created a topic today, I saw that it concerns julia. In Actions I saw

<details>

<summary>the following failure:</summary>

```
FAILED pandapower/test/opf/test_TNEP.py::test_pm_tnep_cigre_dc - julia.core.JuliaError: Exception 'LoadError: ArgumentError: Package PandaModels does not have PowerModels in its dependencies:
- You may have a partially installed environment. Try `Pkg.instantiate()`
  to ensure all packages in the environment are installed.
- Or, if you have PandaModels checked out for development and have
  added PowerModels as a dependency but haven't updated your primary
  environment's manifest file, try `Pkg.resolve()`.
- Otherwise you may need to report an issue with PandaModels
in expression starting at /home/runner/.julia/packages/PandaModels/H0sdB/src/PandaModels.jl:1' occurred while calling julia code:
using PandaModels
FAILED pandapower/test/opf/test_TNEP.py::test_pm_tnep_cigre_ac_S - julia.core.JuliaError: Exception 'UndefVarError: `run_powermodels_tnep` not defined' occurred while calling julia code:
run_powermodels_tnep(buffer_file)
FAILED pandapower/test/opf/test_pandamodels_runpm.py::test_compare_pwl_and_poly - pandapower.auxiliary.OPFNotConverged: PowerModels.jl OPF not converged
FAILED pandapower/test/opf/test_pandamodels_runpm.py::test_without_ext_grid - pandapower.auxiliary.OPFNotConverged: PowerModels.jl OPF not converged
FAILED pandapower/test/opf/test_pandamodels_runpm.py::test_storage_opt - KeyError: 'bus_dc'
FAILED pandapower/test/opf/test_pandamodels_runpm.py::test_runpm_multi_vstab - KeyError: 'bus_dc'
FAILED pandapower/test/opf/test_pandamodels_runpm.py::test_runpm_qflex_and_multi_qflex - KeyError: 'bus_dc'
```

</details>

I have the same thing locally in tests, except for the lines with `FAILED pandapower/test/opf/test_TNEP.py.`

yasirroni, 2024-10-10:
Hi @quant12345, I finally success installing `PandaModels`. Now, I'm also stuck with this same issue, that is `bus_dc`.

SimonUlm, 2024-10-11:
I think this is caused by the commit fcf7bd7ea659c86d7c95b8eb61a7aaee786d3b4b. This also makes the [SimBench](https://github.com/e2nIEE/simbench) plugin unusable.

yasirroni, 2024-10-16:
Change to stable branch on master, test passed. Thanks @SimonUlm.

quant12345, 2024-11-22:
@vogt31337 I saw that in PR #2455 tests for julia (3.9, 2) pass. Need to ask @mfranz13 if there will be any more fixes in the code?

mfranz13, 2024-11-22:
There was some unnecessary key dropping from the net object in the result extraction from PandaModels. Thanks for mentioning this issue to me.

closed 2024-12-09 by vogt31337