e2nIEE/pandapower #2421: time series with multiple loads

WajihaNazHasan, 2024-10-18:
### Feature Checklist

- [ ] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hello!

I am very new to pandapower. I would like to run time series for my designed LV distribution network. I have a total of 12 loads (4 loads are of same 'p_mw' and 'q_mvar' values and and remaining 8 loads are of same value). I have one residential load curve that I would like to apply on all the loads. I have the values 'Time' and 'pu' of load curve that I need to put in in my code. Could you please tell me how to do that because I cannot find any such help. Also please tell me that do I need to create just one controller for all the loads or do I need to create 12 controllers (one for each load)

when I run my code with one load and one controller it works fine but when I introduce any other load it produces this error 'ValueError: Must have equal len keys and value when setting with an iterable'.

Also please can you tell me how to create multiple controllers for the load or how to fix this issue?

Thank you so much

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-11-21:
@WajihaNazHasan is this still relevant? Have you looked into the tutorials? There should be everything you need.

ricoeo, 2024-12-04:
hello @WajihaNazHasan, I could try to answer your questions since the last few weeks I have also been working to make the time-series simulation work. First question, I create one controller for every changing parameter, so for one load, two const controllers. I give one similar example:
`def create_datasource_cccontroller_with_wind_turbine_profiles(
    net, generator_profiles, turbine_id
):
    # Create data sources for real and reactive power
    ds_sgen = DFData(generator_profiles)

    # Apply the ConstControl for real power
    ConstControl(
        net,
        element="sgen",
        element_index=net.sgen.index[turbine_id],
        variable="p_mw",
        data_source=ds_sgen,
        profile_name="real_power_mw",
    )

    # Apply the ConstControl for reactive power
    ConstControl(
        net,
        element="sgen",
        element_index=net.sgen.index[turbine_id],
        variable="q_mvar",
        data_source=ds_sgen,
        profile_name="reactive_power_mw",
    )
`
I don't know if it is the best and elegant way, but it works!
For the second question, I think you could try to follow the first solution first, I haven't encountered something like that.

vogt31337, 2025-03-10:
@WajihaNazHasan like @ricoeo pointed out. You need one ConstController for every object and variable (for example sgen and p_mw). But not for every object (which btw is also possible to do). Most complex thing is to have your datasource correctly organized.

The scond question is related to that, you are setting more (or less) values then for example loads you have given to the constcontroller. For example your datasource has 10 columns, but you only set 5 loads. Or you want to use data from 5 columns for 10 loads.

closed 2025-03-10 by vogt31337