e2nIEE/pandapower #2430: [bug] replace old net.bus_geodata, net.line_geodata by up-to-date net.bus.geo, net.line.geo

SteffenMeinecke, 2024-10-30:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
```

### Issue Description and Traceback

codes that still uses net.bus_geodata, net.line_geodata:

* https://github.com/e2nIEE/pandapower/issues/2335
* https://github.com/e2nIEE/pandapower/issues/2429
* https://github.com/e2nIEE/pandapower/blob/ae39c69b026e5593813557c159eda58c936a6b67/pandapower/protection/utility_functions.py#L367 and following
* https://github.com/e2nIEE/pandapower/blob/ae39c69b026e5593813557c159eda58c936a6b67/pandapower/converter/powerfactory/pp_import_functions.py#L3327 and many other lines in this file

### Expected Behavior

codes should use `net.bus.geo` and `net.line.geo` instead of `net.bus_geodata` and `net.line_geodata`.

### Installed Versions

commit : d9cdd2ee5a58015ef6f4d15c7226110c9aab8140
python : 3.10.11.final.0
python-bits : 64
OS : Windows
OS-release : 10
Version : 10.0.19045
machine : AMD64
processor : AMD64 Family 25 Model 80 Stepping 0, AuthenticAMD
byteorder : little
LC_ALL : None
LANG : de_DE.UTF-8
LOCALE : de_DE.cp1252

pandas : 2.2.2
numpy : 1.26.4
pytz : 2024.1
dateutil : 2.9.0
setuptools : 70.0.0
pip : 24.2
Cython : 0.29.33
pytest : 8.2.1
hypothesis : None
sphinx : 7.3.7
blosc : None
feather : None
xlsxwriter : 3.2.0
lxml.etree : 5.2.1
html5lib : None
pymysql : None
psycopg2 : 2.9.9
jinja2 : 3.1.4
IPython : 8.13.2
pandas_datareader : None
adbc-driver-postgresql: None
adbc-driver-sqlite : None
bs4 : 4.12.3
bottleneck : None
dataframe-api-compat : None
fastparquet : None
fsspec : None
gcsfs : None
matplotlib : 3.7.1
numba : 0.59.0
numexpr : 2.9.0
odfpy : None
openpyxl : 3.1.2
pandas_gbq : None
pyarrow : 15.0.2
pyreadstat : None
python-calamine : None
pyxlsb : None
s3fs : None
scipy : 1.12.0
sqlalchemy : None
tables : 3.9.2
tabulate : 0.9.0
xarray : None
xlrd : 2.0.1
zstandard : 0.22.0
tzdata : 2024.1
qtpy : 2.4.1
pyqt5 : None

### Label

- [X] Relevant labels are selected

vogt31337, 2024-11-21:
@SteffenMeinecke, can this issue be closed?

SteffenMeinecke, 2024-11-25:
> @SteffenMeinecke, can this issue be closed?

I don't think so. There are two open issues in the above list at the moment.

PR #2460 (2024-12-09): Issues/2430
closes #2430
closed 2024-12-09 by vogt31337