e2nIEE/pandapower #2436: [issue] Measurement defined in pp-grid does not exist in ppci, will be deleted!

258796535, 2024-11-03:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
def state_estimation(self, request):
        bb_case_id = request.GET.get("bb_case_id")
        net_json = bb_service.get_json_format_net(bb_case_id=bb_case_id)
        net = pp.from_json_string(net_json)
        net2 = pp.from_json_string(net_json)
        # 假设 df 是你的 DataFrame 对象
        v_stddev = 0.01  # pu
        pq_stddev = 0.01  # MW/Mvar
        i_stddev = 0.002  # kA
        pp.set_user_pf_options(net, init="flat")
        pp.runpp(net, max_iteration=10, tolerance_mva=1e-5, numba=True)
        # for index, row in net.bus.iterrows():
        #     if row['dt_vn_kv'] is None or row['dt_vn_kv'] <0.1:
        #         print(str(index)+':'+row['name']+':'+row['bus_name'])
        #         pp.create_measurement(net, "v", "bus", 1.02, 0.004, index)
        #     else:
        #         pp.create_measurement(net, "v", "bus",row['dt_vn_kv']/row['vn_kv'], 0.004, index)
        #
        # success = est.estimate(net, init="flat")
        # print(success)
        measurements = {
            'bus': net.res_bus.index.tolist(),  # 使用 Bus 的索引
            'value': net.res_bus.vm_pu.tolist(),  # 使用计算得到的电压幅值
            'type': ['voltage'] * len(net.res_bus)  # 假设所有测量都是电压
        }
        print("测量数据:")
        for bus, value in zip(measurements['bus'], measurements['value']):
            print(f"Bus {bus}: {value} p.u.")

        i = 0
        for busIndex in net.bus.index:
            print(busIndex)
            vn_pu = net.res_bus.vm_pu[busIndex]
            pp.create_measurement(net2, "v", "bus", vn_pu, v_stddev, element=busIndex)

            p_mw = net.res_bus.p_mw[busIndex]
            if p_mw != 0:
                pp.create_measurement(net2, "p", "bus", p_mw, pq_stddev, element=busIndex)
            q_mvar = net.res_bus.q_mvar[busIndex]
            if q_mvar != 0:
                pp.create_measurement(net2, "q", "bus", q_mvar, pq_stddev, element=busIndex)

        # line`s
        for lineIndex in net.line.index:
            p_from_mw = net.res_line.p_from_mw[lineIndex]
            pp.create_measurement(net2, "p", "line", p_from_mw, pq_stddev, element=lineIndex, side="from")
            p_to_mw = net.res_line.p_to_mw[lineIndex]
            pp.create_measurement(net2, "p", "line", p_to_mw, pq_stddev, element=lineIndex, side="to")

            q_from_mvar = net.res_line.q_from_mvar[lineIndex]
            pp.create_measurement(net2, "q", "line", q_from_mvar, pq_stddev, element=lineIndex, side="from")
            q_to_mvar = net.res_line.q_to_mvar[lineIndex]
            pp.create_measurement(net2, "q", "line", q_to_mvar, pq_stddev, element=lineIndex, side="to")

        # trafo`s
[... 468 lines cut ...]
4958  None                p         line      508  7.373850e-06     0.01  from
4959  None                p         line      508 -1.833347e-12     0.01    to
4960  None                q         line      508 -1.963852e+00     0.01  from
4961  None                q         line      508  2.646477e-14     0.01    to
4962  None                p         line      509  5.359618e+00     0.01  from
4963  None                p         line      509 -5.332041e+00     0.01    to
4964  None                q         line      509 -4.004684e+01     0.01  from
4965  None                q         line      509  3.492601e+01     0.01    to
4966  None                p         line      510  3.693556e+01     0.01  from
4967  None                p         line      510 -3.693000e+01     0.01    to
4968  None                q         line      510 -2.789059e+01     0.01  from
4969  None                q         line      510  1.178500e+01     0.01    to
4970  None                p         line      511 -1.774857e+01     0.01  fr

### Expected Behavior

nothing

### Installed Versions

* python version: * pandas version:  * networkx version:  * scipy version:  * numpy version:  * packaging version:  * tqdm version:  * deepdiff version:  * Operating System name/version:2.14.11

### Label

- [X] Relevant labels are selected

vogt31337, 2024-11-21:
@258796535, could you provide more information, what you actually want to achieve? The state estimation part will get a big update in pandapower 3.0.0

marcopau, 2025-05-22:
@258796535 Out of bound errors for measurements not existing in ppci were fixed with pandapower 3.0 release. Please double check if you still have any problem.

closed 2025-05-22 by marcopau