e2nIEE/pandapower #2438: [bug] run_contingency_ls2g doesn't work when there are redundant buses

awarsewa, 2024-11-05:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower.networks as pn
import pandapower.contingency as pc
import pandapower as pp 

net = pn.case30()
# Create a redundant bus
pp.create_bus(net, vn_kv=135, name="31", index=30, zone="3.0", max_vm_pu =1.05, min_vm_pu=0.95) # create a new bus
pp.create_switch(net, bus=30, element=29, et="b", type="CB", closed=True) # create a new switch

# Try running contingency analysis
nminus1_cases = {
    "line": {"index": net.line.index.values},
    "gen": {"index": net.gen.index.values}
}
# Running contingency analysis here should fail
pc.run_contingency_ls2g(net, nminus1_cases)
```

### Issue Description and Traceback

When there is a redundant bus, the size of Vinit is not equal to the number of buses anymore. 
I had originally reported this in [issue#2361](https://github.com/e2nIEE/pandapower/issues/2361) and finally wrote an example that reproduces this behaviour.

Traceback (most recent call last):
  File "c:\Users\XXX\Python\pandapower\test_pp_ls2g.py", line 14, in <module>
    pc.run_contingency_ls2g(net, nminus1_cases)
  File "C:\Users\XXX\anaconda3\envs\pandapower\Lib\site-packages\pandapower\contingency\contingency.py", line 225, in run_contingency_ls2g
    s.compute(v_init, net._options["max_iteration"], net._options["tolerance_mva"])       
RuntimeError: SecurityAnalysis::compute: Size of the Vinit should be the same as the total number of buses. Currently:  Vinit: 30 and there are 31 buses.(fyi: Components of Vinit corresponding to deactivated bus will be ignored anyway, so you can put whatever you want there).

### Expected Behavior

run_contingency_ls2g should be able to handle redundant buses e. g. by:
```python
v_init = net._ppc["internal"]["V"][net._pd2ppc_lookups["bus"]]
```

### Installed Versions

INSTALLED VERSIONS
------------------
commit                : 0691c5cf90477d3503834d983f69350f250a6ff7
python                : 3.12.7
python-bits           : 64
OS                    : Windows
OS-release            : 11
Version               : 10.0.22631
machine               : AMD64
processor             : Intel64 Family 6 Model 140 Stepping 1, GenuineIntel
byteorder             : little
LC_ALL                : None
LANG                  : en_US.UTF-8
LOCALE                : English_Australia.1252

pandas                : 2.2.3
numpy                 : 1.26.4
pytz                  : 2024.2
dateutil              : 2.9.0.post0
pip                   : 24.3.1
Cython                : None
sphinx                : None
IPython               : None
adbc-driver-postgresql: None
adbc-driver-sqlite    : None
bs4                   : None
blosc                 : None
bottleneck            : None
dataframe-api-compat  : None
fastparquet           : None
fsspec                : None
html5lib              : None
hypothesis            : None
gcsfs                 : None
jinja2                : None
lxml.etree            : None
matplotlib            : None
numba                 : None
numexpr               : None
odfpy                 : None
openpyxl              : None
pandas_gbq            : None
psycopg2              : None
pymysql               : None
pyarrow               : None
pyreadstat            : None
pytest                : None
python-calamine       : None
pyxlsb                : None
s3fs                  : None
scipy                 : 1.13.1
sqlalchemy            : None
tables                : None
tabulate              : None
xarray                : None
xlrd                  : None
xlsxwriter            : None
zstandard             : None
tzdata                : 2024.2
qtpy                  : None
pyqt5                 : None

### Label

- [X] Relevant labels are selected

awarsewa, 2025-04-01:
@vogt31337 yes that's right. But considering you just put out a major release, I don't know if this is still relevant?

vogt31337, 2025-04-02:
@awarsewa you started the issue :) hopefully it is solved with the major release. 
But it is an important issue, therefore I would Like to keep it Open, until it is resolved.

awarsewa, 2025-04-03:
@vogt31337 I looked into this some more. In init_ls2g() it says that switches are not supported. So my example wouldn't work with ls2g anyways. Even though the v_init issue is fixed by the line of code I posted above, the results will be all NaN and you just run into another error a few lines later. 

I think this one can be closed as it is for a use case that is not officially supported (switches). Replacing switches by very low impedance lines fixes this, because there won't be any redundant buses. If you can think of another way to create redundant buses that does not involve switches, this might still be relevant. Also, if you want to support switches at some point.

vogt31337, 2025-04-04:
@awarsewa you describe a Problem we also Had in Pandapower, there are functions in it which will remove the switches and fuse them together. Since we figured that small impedances can negatively influence the results quite heavily.

I think rbolgaryn wrote some functions which uses internal functions to Setup the ls2g. But I found that ls2g doesn't Support 3W Trafos and ext Ward and stuff (at least the Last Time I checked).

closed 2026-11-11 by vogt31337