e2nIEE/pandapower #2442: [bug] 'dgstrf info' in stdout

DEUCE1957, 2024-11-07:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
In Scipy <1.14.1 (Pandapower currently uses 1.13.1) whenever we solve a Sparse matrix that is singular the following text is sent to stdout:

dgstrf info 1
dgstrf info 1
dgstrf info 1
dgstrf info 1
zgstrf info 2
dgstrf info 1
dgstrf info 1
dgstrf info 1
dgstrf info 1
dgstrf info 1
```
Since this comes from Scipy internally (compiled in C) there is no way to suppress it.

Thankfully this has been fixed in Scipy 1.14.1, see Issue [20933](https://github.com/scipy/scipy/issues/20993)
```

### Issue Description and Traceback

If possible, can PandaPower upgrade to using Scipy 1.14.1?

### Expected Behavior

Expect no messages to be written directly to stdout when encountering a singular sparse matrix.

### Installed Versions

* Python: 1.12.7
* PandaPower: 2.14.11

### Label

- [ ] Relevant labels are selected

vogt31337, 2024-11-21:
Good point, I'll add it to the list, but I don't consider this a bug, merely a nuissance.

vogt31337, 2025-03-10:
scipy version is now updated.

closed 2025-03-10 by vogt31337