e2nIEE/pandapower #2444: [bug] YOUR TITLE

AnkurArohi, 2024-11-08:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
Simply use a EQ_BD file together with other profiles in cim importer ,
compare the no. and attributes of switches with and without the EQ_BD file importer
```

### Issue Description and Traceback

Pandapower has problem in cim import if EQ_BD file is also used

Something went wrong at switches, seems like that terminals for connection with connectivity nodes are missing!

The switch with RDF ID _03ac8 has 3 Terminals!
The switch data: 
     index                                  rdfId  \
504  11799  _03ac83  
505  11797  _03ac83bb   
506  11798  _03ac83bb-6 

                                 name  \
504 3 Ltg  1
505 3 Ltg   2
506 3 Ltg   3

The modelled network has no switch anymore.

If there is no EQ_BD used then it works.

### Expected Behavior

The switches count and attributes should remain same with and without EQ_BD import

### Installed Versions

* python version: 3.12  *

### Label

- [ ] Relevant labels are selected

mrifraunhofer, 2024-11-22:
Hey Ankur,
can you check if you have duplicated ConnectivityNodes in the bus table when you use the EQ_BD. It seems that there is an issue with the EQ_BD.

mrifraunhofer, 2024-11-25:
Hey Ankur, can you provide us an example file? Or reproduceable code? However, a switch may only have 2 Terminals, never 3! Can you check if you have a valid CGMES export? Validating the CGMES input data is not part of our importer. 
Without vaild example files it is difficult investigate the problem.

AnkurArohi, 2024-12-04:
I cannot provide any example unfortunately because these are all proprietary CGMES files, but i can assure you that all these CGMES imports functions for powsybl and GridCal

closed 2024-12-16 by heckstrahler

heckstrahler, 2024-12-16:
Without exmaple data, this issue cannot be pursed further.