e2nIEE/pandapower #2445: [bug] Failing power flow calculation if non mandatory attributes not set

AnkurArohi, 2024-11-08:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
Simply create a network and do not provide "tap_step_degree" for trafo dfand "shift_mv" and "shift_lv_degree" for trafo3w df. Try to run power flow , this results in error of missing shift_mv_degree exception.
```

### Issue Description and Traceback

 After setting PU gens as PQ gens in network, Load Flow did not converge: 'DataFrame' object has no attribute 'shift_mv_degree' 
dc powerflow did not work: 'DataFrame' object has no attribute 'shift_mv_degree' 

### Expected Behavior

Since these columns are not mandatory in the pandapower documentation for load flow calculation this should never occur

### Installed Versions

* python version: 3.12* 

### Label

- [ ] Relevant labels are selected

bott-j, 2024-11-22:
@AnkurArohi, was this after changing to using static generators in the network model? 

Did it fail both normal power flow and dc power flow, and do you have an example of the network being modelled?

AnkurArohi, 2024-11-22:
Yes and it fails for both

bott-j, 2024-11-23:
@AnkurArohi , I haven't been able to reproduce this. 

Was create_transformer_from_parameters() and create_sgen() used to create the equipment in the network? What was the slack/reference bus for the model?

AnkurArohi, 2024-11-23:
Yes it was latest,slack and all other details were same and it works with the old pp version

vogt31337, 2025-03-28:
closed, since transfomer were changed in pandapower 3.0.0

closed 2025-03-28 by vogt31337