e2nIEE/pandapower #2446: Base impedance/admittance of per unit system of admittance matrix

Sciemon, 2024-11-11:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

I am trying to understand the entries of the admittance matrix (net._ppc["internal"]["Ybus"]) as illustrated for example in the [internal data structure tutorial](https://github.com/e2nIEE/pandapower/blob/develop/tutorials/internal_datastructure.ipynb), but I cannot figure out the correct base impedance/admittance for the per unit conversion.

I would assume that an entry in the matrix that represents a line between two buses could be calculated as
```python
Z_base = vn_kv**2 / net._ppc["baseMVA"]
Z = (r_ohm_per_km + x_ohm_per_km*j)*length_km 
z = Z/Z_base
y = 1/z
``` 
where `vn_kv` is the nominal voltage of the from bus and r_ohm_per_km and x_ohm_per_km are parameters of the line.
 
But the results don't match the non diagonal entries of line branches in Ybus (other branch elements I have not checked). Could you point me to what I am missing? What would be the correct `Z_base `?

Thanks!

### Label

- [ ] Relevant labels are selected

pawellytaev, 2024-11-14:
Hi @Sciemon, in general I agree, the entry on the non-diagonal entries on the Ybus matrix should be the negative line admittance connecting the two buses. Do you have a minimal example where your see the inconsitency? Do the lines in your model have `parallel>1`?

FYI, here is where we calculate the base impedance and branch impedance in p.u.: https://github.com/e2nIEE/pandapower/blob/d9ada108b1ac144f58ce7a8b7d8c093c9a9d0ec7/pandapower/build_branch.py#L203

pawellytaev, 2024-11-14:
... and here's where we calculate the Ybus matrix entries:
https://github.com/e2nIEE/pandapower/blob/d9ada108b1ac144f58ce7a8b7d8c093c9a9d0ec7/pandapower/pypower/makeYbus.py#L46

https://github.com/e2nIEE/pandapower/blob/d9ada108b1ac144f58ce7a8b7d8c093c9a9d0ec7/pandapower/pypower/makeYbus.py#L87

Sciemon, 2024-11-26:
@pawellytaev many thanks for the links and the clarification. I had a careless error in my code and could validate the admittance matrix entries after fixing that.