e2nIEE/pandapower #2458: [feature] Improved error handling

vogt31337, 2024-11-22:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [X] Adding new functionality to pandapower

- [ ] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

In pandapower when a bug happens we normally get a stacktrace. But most often they are are hard to understand and are hard debug.

### Feature Description

Maybe we can improve the situation and throw "better" error messages allowing for easier debugging, for example if an operation on a pandas table fails, return where and which table the error happened on.

### Label

- [ ] Relevant labels are selected

pawellytaev, 2024-12-03:
I agree! Also in case there is an issue with any element in the grid that causes an error, pandapower should point to the element (e.g. error in ppc[branch] can be a trafo, trafo3w, line, impedance etc.).

Also, how about extending/updating the pp.diagnostic()? Might be reasonable to update it with possible issues from new elements that came into pandapower (belongs in a separate issue)

vogt31337, 2025-01-08:
Also include Ideas and Stuff from #786