e2nIEE/pandapower #2474: [bug] MemoryError from makeYbus_tcsc - does NOT occur in pandapower version 2.8!

marhofmann, 2024-12-02:
### Bug report checklis

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [X] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [X] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
import pandapower.networks as pn
from pandapower.toolbox import merge_nets

net = pn.case9241pegase()
print(net)
net = merge_nets(net, net, return_net2_reindex_lookup=False)
net = merge_nets(net, net, return_net2_reindex_lookup=False)
net = merge_nets(net, net, return_net2_reindex_lookup=False)
print(net)
pp.runpp(net)
```

### Issue Description and Traceback

I failed to run a power flow with an old grid model using pandapower version 2.14. To reproduce the issue without the original grid data, I merged the 'case9241pegase' multiple times with 'merge_nets'. 
```
This pandapower network includes the following parameter tables:
   - bus (9241 elements)
   - load (4461 elements)
   - sgen (434 elements)
   - gen (1444 elements)
   - shunt (7327 elements)
   - ext_grid (1 element)
   - line (13797 elements)
   - trafo (2252 elements)
   - poly_cost (1445 elements)
   - bus_geodata (9241 elements)
Traceback (most recent call last):
  File "test.py", line 13, in <module>
    net = merge_nets(net, net, return_net2_reindex_lookup=False)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\pandapower\toolbox\grid_modification.py", line 195, in merge_nets
    return _merge_nets(net1, net2, validate=validate, merge_results=merge_results, tol=tol,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\pandapower\toolbox\grid_modification.py", line 251, in _merge_nets
    runpp(net, **runpp_kwargs)
  File "\pandapower\run.py", line 250, in runpp
    _powerflow(net, **kwargs)
  File "\pandapower\powerflow.py", line 84, in _powerflow
    result = _run_pf_algorithm(ppci, net["_options"], **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\pandapower\powerflow.py", line 166, in _run_pf_algorithm
    result = _run_newton_raphson_pf(ppci, options)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\pandapower\pf\run_newton_raphson_pf.py", line 61, in _run_newton_raphson_pf
    ppci, success, iterations = _run_ac_pf_without_qlims_enforced(ppci, options)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\pandapower\pf\run_newton_raphson_pf.py", line 171, in _run_ac_pf_without_qlims_enforced
    V, success, iterations, J, Vm_it, Va_it, r_theta_kelvin_per_mw, T = newtonpf(Ybus, Sbus, V0, ref, pv, pq, ppci, options, makeYbus)
                                                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\pandapower\pypower\newtonpf.py", line 153, in newtonpf
    Ybus_tcsc = makeYbus_tcsc(Ybus, x_control_tcsc, tcsc_x_l_pu, tcsc_x_cvar_pu, tcsc_fb, tcsc_tb)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\pandapower\pf\makeYbus_facts.py", line 18, in makeYbus_tcsc
    Ybus_tcsc = np.zeros(Ybus.shape, dtype=np.complex128)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
numpy.core._exceptions._ArrayMemoryError: Unable to allocate 81.4 GiB for an array with shape (73928, 73928) and data type complex128
```

### Expected Behavior

But if I downgrade to:
* python version: 3.9 
* pandapower: 2.8
* pandas: 1.4
* numba: 0.58
* scipy:  1.13.1
* numpy:  1.24
The power flow runs successfully!

```
This pandapower network includes the following parameter tables:
   - bus (73928 elements)
   - load (35688 elements)
   - sgen (3472 elements)
   - gen (11552 elements)
   - shunt (58616 elements)
   - ext_grid (8 elements)
   - line (110376 elements)
   - trafo (18016 elements)
   - poly_cost (11560 elements)
   - bus_geodata (73928 elements)
 and the following results tables:
   - res_bus (73928 elements)
   - res_line (110376 elements)
   - res_trafo (18016 elements)
   - res_ext_grid (8 elements)
   - res_load (35688 elements)
   - res_sgen (3472 elements)
   - res_shunt (58616 elements)
   - res_gen (11552 elements)
```

### Installed Versions

* python version: 3.11 
* pandapower version: 2.14.11
* pandas version:  2.2.3
* numba version: 0.60
* scipy version:  1.13.1
* numpy version:  1.26.4

### Label

- [X] Relevant labels are selected

vogt31337, 2024-12-02:
@marhofmann, as you can see in the error message pp needs more ram. 81.4 GiB to be exactly. Can you by any chance upgrade to the latest development version of pandapower? Since there the tcsc part was overhauled and changed.

marhofmann, 2024-12-02:
The power flow runs with the latest development version of pandapower for the test gird of the reproducible example!
Also the grid model that I want to simulate will now be processed. However, I tested it with “convert=False” because converting the geodata into a geojson took a long time and I had to interrupt it!

vogt31337, 2024-12-06:
Okay, then I'll close this issue. As it will be fixed with pp 3.0.0

closed 2024-12-06 by vogt31337