e2nIEE/pandapower #2476: t.run_all_tests() AssertionError

Knurris, 2024-12-04:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hi, I get an AssertionError when running pandapower.test.run_all_tests(), what can i do to remedy this error?
[image]

### Label

- [ ] Relevant labels are selected

pawellytaev, 2024-12-04:
Hey @Knurris, which pandapower version are you working on? have you tried the latest develop version?

Knurris, 2024-12-04:
pandapower  2.14.11, is that the latest?

pawellytaev, 2024-12-04:
Yup, that's the latest stable version.

And I could replicate this problem and have the same issue. This is solved on the development version, which will be released soon. So if you clone it and install it (as described on https://www.pandapower.org/start/) this issue shouldn't persist. 

Also, in case you don't work with geodataframes that you want to export in json files, this test is also not so critical

closed 2024-12-05 by Knurris