e2nIEE/pandapower #2492: Provision of Resource file for modeling of FACTS devices

HafizRiaz, 2024-12-16:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [ ] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

I am working on modeling FACTS  devices, so I need the source from where the pandapower model has been built. In the documentation, it has been mentioned that "We implement the FACTS devices based on the following source:
A. Panosyan, “Modeling of advanced power transmission system controllers”, Ph.D. dissertation, Gottfried Wilhelm Leibniz Universität Hannover, 2010." After searching the web, I could not find the source except the paid link of Amazon. if you could share the source, I would be able to search for the upgradation of models, or you may suggest some other rooms or sources, regarding this area. 

### Label

- [X] Relevant labels are selected

KS-HTK, 2024-12-18:
@pawellytaev

pawellytaev, 2024-12-18:
Hey @HafizRiaz please reach out to info@pandapower.org, and then I can send you the document via e-mail there

closed 2024-12-18 by pawellytaev