e2nIEE/pandapower #2507: Usage of trafo vector_group unclear, information inconsistent

heckstrahler, 2025-01-09:
### Feature Checklist

- [X] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [X] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [X] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

vector_group defines the type of a transformer winding:
For example: Yyn, YYY
Upper case for hv side and lower case for lv side.
The definition in the documentation for trafo2w is unclear and for trafo3w it does not exist.
(CGMES for example has more types available then are listed in the pandapower docu)

In literature the vector_group often also contains a number for the shift_degree.
For example: Dzn5, YN0yn0yn0

Example networks in pandapower like lv_schutterwald.json do contain these numbers in their vector_group table.

It is unclear if this is correct grid modeling or leftovers from an older pandapower version.

The shift_degree column is the value that seems to be used for calculations.

In all the documentation should clearly state how the vector groups can look like and internally only one representation should be used (actual calculation code and example networks).

Issue #1580 might be related.

### Label

- [ ] Relevant labels are selected

vogt31337, 2025-03-13:
I am currently in the process of acquiring the following norm: [https://www.vde-verlag.de/normen/1500287/e-din-vde-v-0532-508-2513-vde-v-0532-508-2513-2023-12.html](0532-508-2513-2023-12). I hope it will clarify these things and we can make pandapower compatible according to that norm.

vogt31337, 2025-03-28:
Sadly this norm was not the correct one. It seems that this norm is the one describing everything: https://webstore.ansi.org/standards/iec/iec60076powertransformers