e2nIEE/pandapower #2517: [bug] CIM16 CGMES import error busbar_id and busbar_name is dropped

bp50hz, 2025-01-30:
### Bug report checklis

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [x] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
testcase is missing in:
https://github.com/e2nIEE/pandapower/blob/8009cb53287c1f79e009d3093b1baf5694da17db/pandapower/test/converter/test_from_cim.py
```

### Issue Description and Traceback

https://github.com/e2nIEE/pandapower/blob/develop/pandapower/converter/cim/cim2pp/build_pp_net.py
line 65 in build_pp_net.py:
```
self.net[pp_type] = pd.concat([self.net[pp_type],
        input_df[list(set(self.net[pp_type].columns).intersection(input_df.columns))]],
        ignore_index=True, sort=False)
```

during runtime:
```
input_df.columns = Index(['origin_id', 'name', 'description', 'ConnectivityNodeContainer',
       'origin_profile', 'origin_class', 'ConnectivityNodeContainer_id',
       'Substation', 'zone', 'Substation_id', 'vn_kv', 'ConnectivityNode',
       'cim_topnode', 'busbar_id', 'busbar_name', 'GeographicalRegion_id',
       'GeographicalRegion_name', 'SubGeographicalRegion_id',
       'SubGeographicalRegion_name', 'in_service', 'type'],
      dtype='object')
```

```
self.net[pp_type].columns = 
Index(['name', 'vn_kv', 'type', 'zone', 'in_service', 'geo', 'origin_id',
       'origin_class', 'origin_profile', 'cim_topnode',
       'ConnectivityNodeContainer_id', 'Substation_id', 'description',
       'Busbar_id', 'Busbar_name', 'GeographicalRegion_id',
       'GeographicalRegion_name', 'SubGeographicalRegion_id',
       'SubGeographicalRegion_name'],
      dtype='object')
```

Busbar_id != busbar_id
Busbar_name != busbar_name
-> busbar info gets dropped 

why is the type hardcoded and not based on the busbarsection? 
https://github.com/e2nIEE/pandapower/blob/develop/pandapower/converter/cim/cim2pp/converter_classes/connectivitynodes/connectivityNodesCim16.py
line 266

### Expected Behavior

Busbar_id == Busbar_id
Busbar_name == Busbar_name

### Installed Versions

pandapower  3.0.0.dev1 8009cb5

### Label

- [ ] Relevant labels are selected

bp50hz, 2025-01-31:
possible hotfix:
https://github.com/e2nIEE/pandapower/blob/develop/pandapower/converter/cim/cim2pp/converter_classes/connectivitynodes/connectivityNodesCim16.py
current code:
```
    def convert_connectivity_nodes_cim16(self):
        time_start = time.time()
        self.logger.info("Start converting ConnectivityNodes / TopologicalNodes.")
        connectivity_nodes, eqssh_terminals = self._prepare_connectivity_nodes_cim16()
```
add after line 24:
`connectivity_nodes = connectivity_nodes.rename(columns={'busbar_id': 'Busbar_id', "busbar_name": 'Busbar_name'})`

mrifraunhofer, 2025-03-19:
Hey, thank you fot the info. The busbars infomartin were dropped. This is fixed in PR: #2560
https://github.com/e2nIEE/pandapower/pull/2560

mrifraunhofer, 2025-04-04:
Hey, however, there was a problem with the last PR. The bug will be fixed with the PR: #2573

mrifraunhofer, 2025-04-04:
PR merged

closed 2025-04-04 by mrifraunhofer