e2nIEE/pandapower #2519: [bug] YOUR TITLE

j0m0k0, 2025-02-03:
### Bug report checklis

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [x] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
Install the latest version of pandapower in Mac OS M2, using pip.
Then run the following code, as mentioned in the documentation [here](https://www.pandapower.org/start/)

import pandapower.test
pandapower.test.run_all_tests()
```

### Issue Description and Traceback

Install the latest version of pandapower in Mac OS M2, using pip.
Then run the following code, as mentioned in the documentation [here](https://www.pandapower.org/start/)
```
import pandapower.test
pandapower.test.run_all_tests()
```

There is one test that fails:
```
============================================================================== short test summary info ===============================================================================
FAILED .venv/lib/python3.12/site-packages/pandapower/test/control/test_tap_dependent_impedance.py::test_characteristic_diagnostic - AssertionError: assert 'The characteristic value of 6.0 at the neutral tap position 0 does not match the value 6.000001' in 'Checking transformer characteristics\ntrafo: found 2...
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! stopping after 1 failures !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
======================================================== 1 failed, 181 passed, 12 skipped, 3 xfailed, 229 warnings in 21.68s =========================================================
```

### Expected Behavior

All tests should pass.

### Installed Versions

* python version: 3.12
* pandas version:  2.2.3
*  networkx version:  3.4.2
*  scipy version:  1.13.1
*  numpy version:  1.26.4
*  packaging version:  2.14.11
*  tqdm version:  4.67.1
*  deepdiff version:  8.1.1
*  Operating System name/version: MacOS 14.7 

### Label

- [ ] Relevant labels are selected

tellitos, 2025-02-05:
I get the same message and while running the sample code from the tutorials I also get different angle values for S apparent power angle. Im very thankful for any help or leads.

[image]
[image]

vogt31337, 2025-02-05:
Hello,

thanks for raising the awareness to this issue, currently there is a large PR in the making which will change some of the inner workings regarding this part of pandapower.

@tellitos, could you perhaps point me to which tutorial this problem arose?

BR

tellitos, 2025-02-05:
> Hello,
> 
> thanks for raising the awareness to this issue, currently there is a large PR in the making which will change some of the inner workings regarding this part of pandapower.
> 
> [@tellitos](https://github.com/tellitos), could you perhaps point me to which tutorial this problem arose?
> 
> BR

@vogt31337  thanks for answering im referring to this Tutorial: Four bus system
`pandapower.networks.simple_four_bus_system()[](https://pandapower.readthedocs.io/en/latest/networks/test.html#pandapower.networks.simple_four_bus_system)`

Kind Regards

closed 2026-11-10 by heckstrahler