e2nIEE/pandapower #2527: [feature] Double-phase-to-ground faults in short circuit

zhtee, 2025-02-14:
### Feature Checklist

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [x] Adding new functionality to pandapower

- [ ] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

In short-circuit calculation, I would like to know why there is only two-phase (phase-to-phase) short-circuits and no double-phase-to-ground faults. Is it possible to add this feature?

### Feature Description

Add double-phase-to-ground faults option.

### Label

- [ ] Relevant labels are selected

vogt31337, 2025-03-13:
@zhtee thanks for mentioning. I've tagged @gourab2009 who can hopefully enlighten in this topic. Also this could be a feature added later this year.

zhtee, 2025-03-13:
Dear @gbanerjeefraunhofer ,

Thank you for your reply. As a beginner, I am very pleased to hear this news. This feature will make my learning smoother. Thank you for your efforts!

Yours sincerely,
zhtee

closed 2025-03-13 by zhtee