e2nIEE/pandapower #2532: [bug] changes for characteristics in shunt/trafo/trafo3w are not considered in stored nets and convert_format()

SteffenMeinecke, 2025-02-19:
### Bug report checklis

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [x] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
provided by an added test in Pull Requst ...
```

### Issue Description and Traceback

If networks stored as (json) files were already updated after the last release, the `format_version` was updated, too. If further updates are made but not integrated to the (json) files, the loaded networks will not include the latest updates, even if these updates would usually be catched by the convert_format() function. This is because `convert_format()` won't do corrections to loaded data of the latest `format_version`.

At the moment, the changes for characteristics in shunt, trafo, and trafo3w are such changes, not integrated into stored (json) files. To show up, I added `test_stored_networks.py`.
To fix the added, failing tests, the networks must be stored again including the latest updates.

@heckstrahler @mrifraunhofer @pawellytaev who is going to do the fixes?

1. Bug: The changes for characteristics are not catched by the convert_format() function. The following copy of the relevant code snippet shows that nothing is fitting included:

```
if Version(str(net.format_version)) < Version("3.0.0"):
    _convert_geo_data(net, elements_to_deserialize)
    _convert_group_element_index(net)
    _convert_trafo_controller_parameter_names(net)
    convert_trafo_pst_logic(net)
```

@heckstrahler @mrifraunhofer @pawellytaev who is going to fix this?

### Expected Behavior

* all mandatory element columns exist in all networks that can be loaded from `pandapower.networks`

### Installed Versions

INSTALLED VERSIONS
------------------
commit                : 0691c5cf90477d3503834d983f69350f250a6ff7
python                : 3.10.11
python-bits           : 64
OS                    : Windows
OS-release            : 10
Version               : 10.0.19045
machine               : AMD64
processor             : AMD64 Family 25 Model 80 Stepping 0, AuthenticAMD
byteorder             : little
LC_ALL                : None
LANG                  : de_DE.UTF-8
LOCALE                : de_DE.cp1252

pandas                : 2.2.3
numpy                 : 1.26.0
pytz                  : 2024.1
dateutil              : 2.9.0
pip                   : 25.0
Cython                : 0.29.33
sphinx                : 7.3.7
IPython               : 8.13.2
adbc-driver-postgresql: None
adbc-driver-sqlite    : None
bs4                   : 4.12.3
blosc                 : None
bottleneck            : None
dataframe-api-compat  : None
fastparquet           : None
fsspec                : None
html5lib              : None
hypothesis            : None
gcsfs                 : None
jinja2                : 3.1.4
lxml.etree            : 5.2.1
matplotlib            : 3.9.1
numba                 : 0.60.0
numexpr               : 2.10.2
odfpy                 : None
openpyxl              : 3.1.2
pandas_gbq            : None
psycopg2              : 2.9.9
pymysql               : None
pyarrow               : 16.1.0
pyreadstat            : None
pytest                : 8.2.1
python-calamine       : None
pyxlsb                : None
s3fs                  : None
scipy                 : 1.15.1
sqlalchemy            : None
tables                : 3.9.2
tabulate              : 0.9.0
xarray                : None
xlrd                  : 2.0.1
xlsxwriter            : 3.2.0
zstandard             : 0.23.0
tzdata                : 2024.1
qtpy                  : 2.4.1
pyqt5                 : None

### Label

- [x] Relevant labels are selected

pawellytaev, 2025-02-19:
@hilbrich could you have a look into this?

panos-xenos, 2025-02-20:
@pawellytaev @SteffenMeinecke the missing columns are only optional, so the networks coming from the json files should work without problems. A user may create a net manually without specifying all these columns. Same with the standard trafo types for example. Is this actually a problem?

cc: @hilbrich

panos-xenos, 2025-02-21:
@vogt31337, as discussed, please see attached mv_oberrhein.json file after going through the convert_format function (which includes the convert_trafo_pst_logic function). Let me know if you need anything else.

[mv_oberrhein.json](https://github.com/user-attachments/files/18907510/mv_oberrhein.json)

hilbrich, 2025-02-24:
Hi @SteffenMeinecke ,

> Dear [@panos-xenos](https://github.com/panos-xenos),
> 
> > I now realize that the issue is that the json files were re-created with v3.0, hence the logic doesn't go into the convert_format functions.
> 
> right, this is the main issue. Despite the characteristic columns are optional, I'd say that it is best to convert (by `convert_format`) the information of 2.14.x version, such as `net.trafo.vk_percent_characteristic`, into the new characteristic parameter columns. (My suggestion - without knowing about what you implemented - is that the characteristic models can be somehow converted into each other).

That is possible, but I see an issue. Are there really testfiles that contain vk_percent_characteristics? The problem that I see is, that this only reflects tap_dependent_impedances, and we would also need to create the information about ratio and shift for every tap position. This is, because in 2.14.x, only tapdependent impedance was considered, but no tables.

SteffenMeinecke, 2025-02-24:
> Are there really testfiles that contain vk_percent_characteristics?

I guess, there was nobody who added a testfile for that.

SteffenMeinecke, 2025-02-28:
In [reconvert_power_system_test_case_jsons](https://github.com/SteffenMeinecke/pandapower/tree/reconvert_power_system_test_case_jsons), I started to reconvert the power system test cases the originates from matpower cases.
json files which must be restored with the new trafo/shunt/trafo3w parameters are:

    "case11_iwamoto",
    "mv_oberrhein",
    "lv_schutterwald",
    "ieee_european_lv_asymmetric"

BDonnot, 2025-03-11:
Hello,

I am writing here because the issue I face is pretty similar. For the grid2op ecosystem, we saved and stored different grids using pandapower (for a few years now) in json format.

Unfortunately, the switch to pandapower 3 breaks lots of things. I don't really know 100% if things are related to this issue but the Ybus matrix i got with pandapower 3 and pandapower 2.xx on the same json file (generated with pandapower 2.xx) are quite different for most of the coefficients were there are trafos.

vogt31337, 2025-03-13:
I reconverted all the networks in pandapower 3.0.0 prior to releasing the final version. I simply forgot to close this issue.
@BDonnot the changes in the Y-Matrix can be related to the rather large change of this characteristics model. I will let this issue remaining open until we figured this out.

BDonnot, 2025-03-13:
Hello,

Thanks for answering. Do you have a systematic way to convert "old" pandapower data to format compatible with the newest version ? If you so, can you share it ?

And, beside that, in https://github.com/e2nIEE/pandapower/issues/2557 I noticed (visually, so I might have missed some changes) that the case118().trafo table is not modified for the `vk_percent` , `vkr_percent`, `i0_percent`or `pfe_kw` attribute.
But the case 118 should not have changed (it has been created 40 or 50 years ago) and if I run a powerflow with pandapower 2 or pandapower 3 it should still compute the same quantities.
I don't really understand all the details, but I would expect that any of the ieee grid lead to different results with pandapower 2 and pandapower 3 (for these grids again :-) )

vogt31337, 2025-04-07:
@BDonnot yes it is the function discussed prior, it is called "convert format" and depending on the format it finds, it will execute different functions to update the network. Maybe just execute whatever you want from there.

Regarding the case118 ieee grid, afaik the modelling is "not good" and it has values in it which are somewhat problematic...
That could explain why the results are different... I am not sure what to do about that, maybe we include the original files and write a converter which get's executed during tests.

vogt31337, 2025-12-16:
@BDonnot , I've fixed now a lot of issues, and I found more problems, with the IEEE case files. The case14 has a generator too much, and the case39 also have some kind of problem. I think we will need to double check all the case files, probably we will reconvert them. The weird errors are stacking up. Also I want now some "simple" checks, if for example the amount of trafos or generators or so are correct.

closed 2026-11-11 by KS-HTK