e2nIEE/pandapower #2535: Missing docstring description in create_gen for slack

kyribaker, 2025-02-20:
### Feature Checklist

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

In create.py, the function create_gen() is missing the description for the "slack" argument within the docstring.

### Label

- [ ] Relevant labels are selected

vogt31337, 2025-03-13:
@kyribaker thanks for bringing this up. Feel free to submit a PR.

vogt31337, 2025-11-24:
@kyribaker, in the latest documentation they are included. Closing becaus it is fixed.