e2nIEE/pandapower #2536: [feature] Type Hints

kornerc, 2025-02-21:
### Feature Checklist

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [x] Adding new functionality to pandapower

- [ ] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

pandapower provides type information in the doc-strings e.g.
https://github.com/e2nIEE/pandapower/blob/01d80e6cae5a3a15f1e1d44f0d3fd014c7135293/pandapower/create.py#L37

It would be helpful to also have [type hints](https://docs.python.org/3.8/library/typing.html) in the code for function parameters and return types.

### Feature Description

Add [PEP484](https://peps.python.org/pep-0484/) type hints to functions etc.
IDEs like VS Code and PyCharm understand these type hints and can support with linting.

E.g.
```python
def create_empty_network(name="", f_hz=50., sn_mva=1, add_stdtypes=True):
    ...
```
becomes
```python
def create_empty_network(name: str = "", f_hz: float = 50., sn_mva: float = 1, add_stdtypes: bool = True) -> pandapowerNet:
    ...
```

Since the smallest Python version currently supported by pandapower is 3.8, it would be needed to stick to type hints compatible with Python 3.8 e.g.
```python
from typing import List
my_list: List[int] = [1, 2, 3]
````
instead of
```python
my_list: list[int] = [1, 2, 3]
````

I think it would be sufficient to add type hints to functions which are most frequently used by users e.g. in `pandapower/pandapower/create.py` for now.

**Is there principal interest in this feature?
If yes I can start making PRs for this.**

It would also be possible to use static type checking with [mypy ](https://www.mypy-lang.org/) to validate the type hints but this is fully optional. 

### Label

- [ ] Relevant labels are selected

Fabsit, 2025-02-21:
> If yes I can start making PRs for this.

it would be highly appreciated! Tag me as reviewer. If possible, work with the pp3 dev release.

kornerc, 2025-02-21:
> it would be highly appreciated! Tag me as reviewer. If possible, work with the pp3 dev release.

Perfect then I will start with the PR.
Two small questions from my side:
- Is it planned to keep the support for Python 3.8 in pp3?
- Which is the `pp3 dev release` branch; is it the branch `develop`?

Fabsit, 2025-02-21:
yes develop. or does it make sense for you to not support python 3.8 and work directly in python_313?
Regarding which python version are supported in which branch, @vogt31337 or @heckstrahler could you please answer this?

vogt31337, 2025-02-21:
Python 3.8 ended Like 4 months ago: https://endoflife.date/python

I would suggest to not officially support it anymore. 
Typing will be A lot of work. I think this will be a Kind of Community effort...

One big question, can you do it smaller pieces? Like maybe File by File? Otherwise Changes in between will lead to a Lot of conflicts...

kornerc, 2025-02-21:
# Python Version
> Python 3.8 ended Like 4 months ago: https://endoflife.date/python
> 
> I would suggest to not officially support it anymore.

In that case I would prefer to use features which aren't available in Python 3.8 and would support Python >= 3.9.

For me both is fine, having the branch `python_313` as base or `develop`.
I can wait for the merge of `python_313` into `develop` if this is planned to me merged soon.

The most relevant type hinting features are in my opinion:

## 3.8
```python
from typing import List, Optional

def foo(l: List[int], o: Optional[int] = None) -> int:
    return sum(l) + (o or 0)
```

## 3.9
```python
from typing import Optional

def foo(l: list[int], o: Optional[int] = None) -> int:
    return sum(l) + (o or 0)
```

## 3.10
```python
def foo(l: list[int], o: int | None = None) -> int:
    return sum(l) + (o or 0)
```

# Split PR size
> One big question, can you do it smaller pieces? Like maybe File by File? Otherwise Changes in between will lead to a Lot of conflicts...

Yes I totally agree.
Packing this in a single PR is 1) lots of work, 2) unlikely to be ever finished, 3) difficult to review, and 4) will result - as you already pointed out-  in many merge conflicts.

I would start with the functions most often used by the users e.g. `pandapower/pandapower/create.py` and make a single PRs per file (or groups of files) which have a digestible size.
Based on that we can also make adjustments in the case we should face certain pitfalls early on.

thomas-whaley-poco, 2025-04-22:
I'm very excited for this change! I have been adding stub files for our pandapower production code so that we can run mypy on it. This would remove that need. If you want any help in this effort I'm more than willing to help 😄

kornerc, 2025-04-22:
> I'm very excited for this change! I have been adding stub files for our pandapower production code so that we can run mypy on it. This would remove that need. If you want any help in this effort I'm more than willing to help 😄

Thanks for reaching out!

My plan is to finish the type hinting of all public functions in `create.py` so that we can merge #2543 (it was a bit more work than I've expected 😄).
Then, type hinting the remaining code base can be made in multiple PRs and can easily be split up.

Help would be highly appreciated!

PR #2543 (2025-07-10): Type Hints the Beginning
Start adding type hints for pandapower and check them with [mypy](https://www.mypy-lang.org/).

Closes #2536

Since adding type hints for the entire pandapower library is infeasible in a single PR, it will focus on the following points.
- [x] Configuration files for checking the type hints (`pyproject.toml` and `mypy.ini`)
  It's in principal possible to have the entire configuration in `pyproject.toml` but separating those helps in keeping `pyproject.toml` compact.
- [x] Run mypy on most of the codebase without errors.`
  mypy checks now most of the codebase, but I've [... cut]