e2nIEE/pandapower #2546: [feature] kwargs handling in controller

mfisch42, 2025-03-06:
### Feature Checklist

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [x] Adding new functionality to pandapower

- [x] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

Hello everyone, 

it's possible to pass kwargs as input in the init of the controller class. The kwargs are not used further or set to any variables. I did not find any documentation in the controller class why it is possible to pass kwargs when they can not be used.
The only documentation I found are in "drop_same_type_existing_controllers" and "log_same_type_existing_controllers", where the following is written:
" ****kwargs** - unused arguments, given to avoid unexpected input arguments "

But the kwargs are also not taken into consideration here.

### Feature Description

My idea is to set the kwargs as attributes of the controller object (self.#### ). The implementation of the following two lines into the init of the controll class would look like this:

  for key, value in kwargs.items():
        setattr(self, key, value)

Then, the init would look like this: 
```
def __init__(self, net, in_service=True, order=0, level=0, index=None, recycle=False,
                 drop_same_existing_ctrl=False, initial_run=True, overwrite=False,
                 matching_params=None, **kwargs):
        super(Controller, self).__init__(net, index, **kwargs)
        self.matching_params = dict() if matching_params is None else matching_params
        # add oneself to net, creating the ['controller'] DataFrame, if necessary
        # even though this code is repeated in JSONSerializableClass, it is necessary because of how drop_same_existing_controller works
        # it is still needed in JSONSerializableClass because it is used for characteristics
        if index is None and "controller" in net.keys():
            index = get_free_id(net.controller)
        self.index = self.add_controller_to_net(net=net, in_service=in_service, initial_run=initial_run,
                                                order=order, level=level, index=index, recycle=recycle,
                                                drop_same_existing_ctrl=drop_same_existing_ctrl,
                                                overwrite=overwrite, matching_params=matching_params, **kwargs)
        # write kwargs in self
        for key, value in kwargs.items():
            setattr(self, key, value)
```

Question: 
@pawellytaev 
Is there anything against setting the kwargs as attributes of the controllers? Are they actually intended for something else? What is your opinion on this idea? :) 

### Label

- [ ] Relevant labels are selected

pawellytaev, 2025-03-06:
I think it's a good idea. This behaviour would be in a way similar to the `kwargs` in the `create_...` functions.

Or we delete the `kwargs` argument completely, to avoid confusion. But this might have issues with backwards compatibility of code.

vogt31337, 2025-03-10:
I would advocate for deleting the `kwargs`, since why should the super class have `kwargs` ? And for a derived controller, In my opinion it can have a `kwargs`, but it should be up to the developer to handle that.
On the other hand I would like to implement a "serilization" interface, which allows for an easier saving / loading of such controllers.

KS-HTK, 2025-03-10:
I would not agree that this is a good Idea, these Attributes would have to be saved together with the network, and they could be anything including functions or complex objects that might be difficult to serialize to any of the offered save options.

I prefer removing ambiguity and not adding it. If a User requires a Controller with extra parameters they should implement a custom controller class, inheriting functionality where applicable.

vogt31337, 2025-03-13:
> I would not agree that this is a good Idea, these Attributes would have to be saved together with the network, and they could be anything including functions or complex objects that might be difficult to serialize to any of the offered save options.

What exactly do you mean? I thought about making an interface which the user has to implement and therefore extra parameter handling needs to be done by the controller implementer.

closed 2025-07-17 by KS-HTK