e2nIEE/pandapower #2552: [bug] [plotting] Colorbar for Line Loading Not Displayed Correctly in pf_res_plotly

TheBigFudge, 2025-03-11:
### Bug report checklis

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [x] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
import pandapower.networks as pn
from pandapower.plotting.plotly import pf_res_plotly

# Create a sample network
net = pn.mv_oberrhein()
pp.runpp(net)

# Plot the results
pf_res_plotly(net, filename='test.html')
```

### Issue Description and Traceback

Description:
I am using pandapower==3.0 and plotly==6.0 to visualize power flow results with pf_res_plotly. However, the colormap legend for line loading does not appear correctly. I have tried different values for cpos_load, but the issue persists. The expected behavior is that the colorbar for line loading should be displayed correctly according to the climits_load range.

Steps to Reproduce:

1. Run a power flow calculation in a Pandapower network.
2. Use pf_res_plotly() to visualize the results.
3. Observe that the line loading colorbar is either missing or incorrectly positioned.

Observed Behavior:

- The colorbar does not appear correctly.
- Changing cpos_load does not fix the issue.
- The bus voltage legend is displayed correctly, but the loading map is missing.

see plot example

[image]

### Expected Behavior

Expected Behavior:
The colorbar for line loading should be displayed at the correct position as defined by cpos_load.

### Installed Versions

I have tested this with pandapower==3.0 and plotly==6.0.

also tested older version of plotly

### Label

- [x] Relevant labels are selected

vogt31337, 2025-03-13:
@TheBigFudge plotly seems to have changed there js api and things are breaking. Could you by any chance try an earlier plotly version and check if this also happens?

TheBigFudge, 2025-03-19:
hey! i have tried it with some plotly 5.X versions, but that didn't help. However, when I switched back to older pandapower version 2.12.x, both colour bars were visible again

vogt31337, 2025-03-21:
I am currently experiencing similar issued, i think the internal conversion Routine does Not Like empty fields. I will probably soon build a Patch and repair this.