e2nIEE/pandapower #2556: [feature] make res_ tables dependent on actual elements in net

vogt31337, 2025-03-13:
### Feature Checklist

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [x] Adding new functionality to pandapower

- [x] Changing existing functionality in pandapower

- [x] Removing existing functionality in pandapower

### Problem Description

Currently in results.py:

```
def get_relevant_elements(mode="pf"):
    if mode == "pf" or mode == "opf":
        return ["bus", "bus_dc", "line", "line_dc", "trafo", "trafo3w", "impedance", "ext_grid",
                "load", "motor", "sgen", "storage", "shunt", "gen", "ward",
                "xward", "dcline", "asymmetric_load", "asymmetric_sgen",
                "switch", "tcsc", "svc", "ssc", "vsc"]
    elif mode == "sc":
        return ["bus", "line", "trafo", "trafo3w", "ext_grid", "gen", "sgen", "switch"]
    elif mode == "se":
        return ["bus", "line", "trafo", "trafo3w", "impedance", "switch", "shunt"]
    elif mode == "pf_3ph":
        return ["bus", "line", "trafo", "ext_grid", "shunt",
                "load", "sgen", "storage", "asymmetric_load", "asymmetric_sgen"]
```

this code generates a lot of unused dataframes.

### Feature Description

Make the generation of the result dataframes dependent on the actual elements in the grid, e.g. net.bus -> net.res_bus.

### Label

- [ ] Relevant labels are selected

ascheidl, 2025-03-17:
what would be the advantage ? is it that faster without creating empty res tables ?

vogt31337, 2025-03-21:
It generates a Lot of unused tables. I was wondering why I have a Motor table in a simple Network. Now I know why.