e2nIEE/pandapower #2557: [bug] Doc not up to date + behaviour mismatch on ieee cases between pp3 and pp2

BDonnot, 2025-03-13:
### Bug report checklis

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [x] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import numpy as np
import pandapower
import pandapower as pp
import pandapower.networks as pn
from importlib.metadata import version
from packaging.version import Version
import warnings

from lightsim2grid.gridmodel import init_from_pandapower
pp_ver_3 = Version(pandapower.__version__) >= Version("3")
tol = 1e-5

if pp_ver_3:
    from pandapower.pypower.idx_brch import F_BUS, T_BUS, BR_R, BR_X, BR_B, BR_G
else:
    from pandapower.pypower.idx_brch import F_BUS, T_BUS, BR_R, BR_X, BR_B
    
def _wye_delta_new(r, x, g, b, r_ratio=None, x_ratio=None):
    """taken from pandapower 3 source code"""
    if r_ratio is None:
        r_ratio = 0.5 + np.zeros_like(r)
    if x_ratio is None:
        x_ratio = 0.5 + np.zeros_like(x)
        
    tidx = (g != 0) | (b != 0)
    za_star = r[tidx] * r_ratio[tidx] + x[tidx] * x_ratio[tidx] * 1j
    zb_star = r[tidx] * (1 - r_ratio[tidx]) + x[tidx] * (1 - x_ratio[tidx]) * 1j
    zc_star = 1 / (g + 1j*b)[tidx]
    zSum_triangle = za_star * zb_star + za_star * zc_star + zb_star * zc_star
    zab_triangle = zSum_triangle / zc_star
    zac_triangle = zSum_triangle / zb_star
    zbc_triangle = zSum_triangle / za_star
    r[tidx] = zab_triangle.real
    x[tidx] = zab_triangle.imag
    yf = 1 / zac_triangle
    yt = 1 / zbc_triangle
    # 2 because in makeYbus Bcf, Bct are divided by 2:
    g[tidx] = yf.real * 2
    b[tidx] = yf.imag * 2
    g_asym = np.zeros_like(g)
    b_asym = np.zeros_like(b)
    g_asym[tidx] = 2 * yt.real - g[tidx]
    b_asym[tidx] = 2 * yt.imag - b[tidx]
    return r, x, g, b

print(f"pandapower version: {version('pandapower')}")
print(f"pandas version: {version('pandas')}")
print(f"numpy version: {version('numpy')}")
print(f"scipy version: {version('scipy')}")
print(f"lightsim2grid version: {version('lightsim2grid')}")

### main part of the code

# https://github.com/e2nIEE/pandapower/issues/2532
CASE = "case118"
[... 160 lines cut ...]

- deepdiff==8.3.0
- LightSim2Grid==0.10.2
- networkx==3.4.2
- numpy==1.26.4
- orderly-set==5.3.0
- packaging==24.2
- pandas==2.2.3
- pybind11==2.13.6
- python-dateutil==2.9.0.post0
- pytz==2025.1
- scipy==1.13.1
- setuptools==76.0.0
- six==1.17.0
- tqdm==4.67.1
- tzdata==2025.1

With two changes (because I compare 2 venvs): 

- pandapower==2.14.11 for the venv with pandapower 2
- pandapower==3.0.0 for the venv with pandapower 3

### Label

- [ ] Relevant labels are selected

vogt31337, 2025-03-13:
@BDonnot thanks for bringing this up. Could you by any chance add the version of lightsim2grid you were using? Since ls2g needed to be patched to be fully compatible.

BDonnot, 2025-03-13:
Yes of course, lightsim2grid is 0.10.2 as shown in the last paragraph :-)

What bothers me here is that the ieee 118 leads to different results (flows and voltages) depending on pandapower version. I am not sure this is desirable.

Also, if you have a script that convert the "old" grid to the new grid then i'd be happy to have it. Because we stored lots of grid in pandapower format for the l2rpn competitions / grid2op eco system and I can't convert them unfortunately :-(

pawellytaev, 2025-03-13:
Hi @BDonnot,
Thanks for bringing this up!
Took a brief look into this and found the reason why there is the mismatch in powerflow results between 2.14.11 and 3.0
**TL;DR**: the `i0_percent` of the trafos is negative in the case118.json. If you multiply it with -1, the results match with ls2g. I added these lines in your script:
```
...
CASE = "case118"
net = pn.case118()
if pp_ver_3:
    net.trafo.i0_percent *= -1
...
```
... which gives me these results:

[image]

**Detailed answer:**
As we split the BR_B (complex) in pp3.0 into BR_B and BR_G, both float, we found some weird formulas in pandapower 2.X when calculating the trafo BR_B (which was in fact `b - 1j*g`). For example, [here](https://github.com/e2nIEE/pandapower/blob/f499d71da1e7d16156911f26102bd46b0737eeff/pandapower/build_branch.py#L378) we saw that if `i0` was smaller than 0, the `b_img` was smaller than 0 and then - if it is the case - set to 0 in the next line and then multiplied with `np.sign(i0)` in line 382 again.

We thought, why is there even a negative `i0_percent`? Technically this doesn't make sense: the no-load current of a trafo will always flow from phase to ground, and not vice-versa. So we fixed the formulas, see [here](https://github.com/e2nIEE/pandapower/blob/5927f5b7c399978a1a38daf4d33d9031da19360e/pandapower/build_branch.py#L558).
We still have the `b_mva_squared[b_mva_squared < 0] = 0` here, but `b_mva_squared` actually cannot be smaller than 0 as the admittance of the trafo shall always be greater or equal to its conductance `y >= g`.

As we changed all of this, we got failing tests in the `from_ppc` converter, which has the purpose of importing pypower ppc "files" into pandapower, which we also changed so we import positive `i0_percent` values, see here: https://github.com/e2nIEE/pandapower/blob/5927f5b7c399978a1a38daf4d33d9031da19360e/pandapower/converter/pypower/from_ppc.py#L272
and apparently we changed only the test files there and not the json files used by `pandapower.networks.caseXYZ()`

We are currently working on the re-conversion of matpower files via `from_ppc` to pandapower. We will then update the corrected caseXYZ.json to the repo. I will be out of office from tomorrow until 25th, so I can continue working on that only from then on.

@BDonnot please remember some trafos will be handled as impedances as pointed out by Roman here: https://github.com/Grid2op/lightsim2grid/issues/88 ... I guess the issue here is also related to your [answer](https://github.com/Grid2op/lightsim2grid/issues/88#issuecomment-2315603009)

As a quick fix in grid2op, you can multiply the `i0_percent` after importing the `grid.json` files with -1, and we'll keep you posted here as soon as we updated the json's in pandapower.

**Summary of To-Do's:**

- [ ] document the BR_B (complex) to BR_G and BR_B (float) changes somewhere or more clear
- [ ] re-convert the caseXYZ.json and match them with matpower/pypower (found a few mismatches between matpower and pandapower)
- [ ] update the trafo documentation accordingly (thanks for pointing it out!)

BR Pawel

pawellytaev, 2025-03-13:
P.S. Although the Z and Y mismatches are better than before, the Y mismatch is still off, can't figure out why right now... will need to comparte the `ppc['branch']` on both versions in detail. Here is the updated script with a csv export and a bit of `pandas.DataFrame.describe()` added:
```
import numpy as np
import pandapower
import pandapower as pp
import pandapower.networks as pn
from importlib.metadata import version
from packaging.version import Version
import warnings
import pandas as pd

from lightsim2grid.gridmodel import init_from_pandapower
pp_ver_3 = Version(pandapower.__version__) >= Version("3")
tol = 1e-5

if pp_ver_3:
    from pandapower.pypower.idx_brch import F_BUS, T_BUS, BR_R, BR_X, BR_B, BR_G
else:
    from pandapower.pypower.idx_brch import F_BUS, T_BUS, BR_R, BR_X, BR_B
    
def _wye_delta_new(r, x, g, b, r_ratio=None, x_ratio=None):
    """taken from pandapower 3 source code"""
    if r_ratio is None:
        r_ratio = 0.5 + np.zeros_like(r)
    if x_ratio is None:
        x_ratio = 0.5 + np.zeros_like(x)
        
    tidx = (g != 0) | (b != 0)
    za_star = r[tidx] * r_ratio[tidx] + x[tidx] * x_ratio[tidx] * 1j
    zb_star = r[tidx] * (1 - r_ratio[tidx]) + x[tidx] * (1 - x_ratio[tidx]) * 1j
    zc_star = 1 / (g + 1j*b)[tidx]
    zSum_triangle = za_star * zb_star + za_star * zc_star + zb_star * zc_star
    zab_triangle = zSum_triangle / zc_star
    zac_triangle = zSum_triangle / zb_star
    zbc_triangle = zSum_triangle / za_star
    r[tidx] = zab_triangle.real
    x[tidx] = zab_triangle.imag
    yf = 1 / zac_triangle
    yt = 1 / zbc_triangle
    # 2 because in makeYbus Bcf, Bct are divided by 2:
    g[tidx] = yf.real * 2
    b[tidx] = yf.imag * 2
    g_asym = np.zeros_like(g)
    b_asym = np.zeros_like(b)
    g_asym[tidx] = 2 * yt.real - g[tidx]
    b_asym[tidx] = 2 * yt.imag - b[tidx]
    return r, x, g, b

print(f"pandapower version: {version('pandapower')}")
print(f"pandas version: {version('pandas')}")
print(f"numpy version: {version('numpy')}")
print(f"scipy version: {version('scipy')}")
print(f"lightsim2grid version: {version('lightsim2grid')}")

### main part of the code

# https://github.com/e2nIEE/pandapower/issues/2532
CASE = "case118"
net = pn.case118()
if pp_ver_3:
    net.trafo.i0_percent *= -1

pp.runpp(net, lightims2grid=False, numba=False)
with warnings.catch_warnings():
    warnings.filterwarnings("ignore")
    gridmodel = init_from_pandapower(net)
    
# fix the "None" in the tap side
net.trafo.loc[np.array([el is None for el in net.trafo["tap_side"]]), "tap_side"] = "hv"

[... 99 lines cut ...]

print("--- describe trafos ---")
print("Trafo vk")
print(net.trafo.vk_percent.describe())
print("Trafo vkr")
print(net.trafo.vkr_percent.describe())
print("Trafo pfe")
print(net.trafo.pfe_kw.describe())
print("Trafo i0")
print(net.trafo.i0_percent.describe())

print("--- describe lines ---")
print("Line R")
print((net.line.r_ohm_per_km * net.line.length_km).describe())
print("Line X")
print((net.line.x_ohm_per_km * net.line.length_km).describe())
print("Line C")
print((net.line.c_nf_per_km * net.line.length_km).describe())

```

I am also attaching the `ppc['branch']` csv's here:

[ppc_branch_pp2.csv](https://github.com/user-attachments/files/19237519/ppc_branch_pp2.csv)
[ppc_branch_pp3.csv](https://github.com/user-attachments/files/19237522/ppc_branch_pp3.csv)

pawellytaev, 2025-03-27:
> I think this is the right thing to do. And, if I may, I posted an "issue" a few months ago (see [#2131](https://github.com/e2nIEE/pandapower/issues/2131)). If you are to reconvert them again, from a user perspective, it would be better if the default grid matches the expected results with the default values of pp.runpp.
> Because right now, if you want to run the case118 like in matpower or in any other software, you need to specify `trafo_model='pi'` because the data have been computed with `trafo_model="pi"` but in pandapower by default it's the "t".

good idea, we can consider that when we re-work the json's

> This bothered me also back in 2020. I was trying to match the results of pandapower in lightsim2grid. I could not explain this piece of code but implemented finally in lightsim2grid without understanding it.
> 
> >

I was digging deeper into this issue and found out a few things more things that I honestly don't understand.

**Comparison between pp and ls2g**
I took a deeper look into the comparison and corrected the code in a way that we compare `g_ls` with `g_comp` and `b_ls` with `b_comp`. 
```python
# extract values computed by pandapower
if pp_ver_3:
    r_comp = net._ppc["branch"][-net.trafo.shape[0]:, BR_R]
    x_comp = net._ppc["branch"][-net.trafo.shape[0]:, BR_X]
    g_comp = net._ppc["branch"][-net.trafo.shape[0]:, BR_G]
    b_comp = net._ppc["branch"][-net.trafo.shape[0]:, BR_B]
else:
    r_comp = net._ppc["branch"][-net.trafo.shape[0]:, BR_R].real
    x_comp = net._ppc["branch"][-net.trafo.shape[0]:, BR_X].real
    g_comp = -net._ppc["branch"][-net.trafo.shape[0]:, BR_B].imag # in pp<3.0, BR_B was b-1j*g
    b_comp = net._ppc["branch"][-net.trafo.shape[0]:, BR_B].real

# extract values computed by lightsim2grid
r_ls = np.array([el.r_pu for el in gridmodel.get_trafos()])
x_ls = np.array([el.x_pu for el in gridmodel.get_trafos()])
h_ls = np.array([el.h_pu for el in gridmodel.get_trafos()])
g_ls = h_ls.real
b_ls = h_ls.imag
```

Comparing them
```python
print(f"Mismatch with lightsim corrected (y): {(np.sqrt(np.square(g_ls) + np.square(b_ls)) - np.sqrt(np.square(g_comp) + np.square(b_comp))).max()}")
print(f"Mismatch with lightsim (g): {(np.abs(g_ls - g_comp)).max()}")
print(f"Mismatch with lightsim (b): {(np.abs(b_ls - b_comp)).max()}")
```
leads to:
pandapower3
Mismatch with lightsim corrected (y): 0.0
Mismatch with lightsim (g): 5.7183194269137816e-18
Mismatch with lightsim (b): 0.0

and
pandapower2
Mismatch with lightsim corrected (y): 0.0
Mismatch with lightsim (g): 2.8617008122986537e-18
Mismatch with lightsim (b): 2.220446049250313e-16

... and if compare the power flow results between pandapower and lightsim2grid, I get the following:
```python
print("Comparison ls2g ac_pf() and pandapower runpp")
V_ls = gridmodel.ac_pf(np.array([1.0]*len(net.bus)), 15, 1e-8)
Vm_ls = np.abs(V_ls)
Va_ls = np.angle(V_ls)

print(f"Vm max mismatch ls2g {lightsim2grid.__version__} vs. pandapower {pp.__version__}")
print((net.res_bus.vm_pu.values - Vm_ls.T).max())

print(f"Va max mismatch ls2g {lightsim2grid.__version__} vs. pandapower {pp.__version__}")
print((net.res_bus.va_degree.values - Va_ls.T).max())
```

pandapower3:
Comparison ls2g ac_pf() and pandapower runpp
Vm max mismatch ls2g 0.10.2 vs. pandapower 3.0.0
9.519096622057077e-11
[... 70 lines cut ...]
branch_df_3 = pd.read_csv('ppc_branch_pp3.csv', index_col=0)
branch_df_2 = pd.read_csv('ppc_branch_pp2.csv', index_col=0)

BR_B3 = branch_df_3['BR_B'].values
BR_G3 = branch_df_3['BR_G'].values

BR_B2 = np.array(branch_df_2['BR_B'].values, dtype=np.complex128).real
BR_G2 = -np.array(branch_df_2['BR_B'].values.imag, dtype=np.complex128).imag

print(BR_B3[-len(net.trafo):]-BR_B2[-len(net.trafo):])
print(BR_G3[-len(net.trafo):]-BR_G2[-len(net.trafo):])
```

leads to:
[-0.         -0.         -0.         -0.         -0.         -0.
 -0.         -1.27600825 -0.         -1.6160267  -0.         -0.08900047
 -0.32800001]
[0.00000000e+00 0.00000000e+00 0.00000000e+00 0.00000000e+00
 0.00000000e+00 0.00000000e+00 0.00000000e+00 1.39716152e-04
 0.00000000e+00 2.83311202e-04 0.00000000e+00 1.39359822e-05
 2.28540095e-06]

I will look deeper into the pandapower code to see where the differences come from. Will keep you posted.

What I don't really understand is why I get the same voltage magnitude after comparing ls2g with pp3.0 and pp2.4... Is there a dependency with pandapower when calling `init_from_pandapower(net)`? At first glance it looks like ls2g is calculating its branch values based on the pandas dataframes, not based on `ppc`, am I right?

pawellytaev, 2025-03-27:
P.S. here's again the final reproducible example
```python
import numpy as np
import pandapower
import pandapower as pp
import pandapower.networks as pn
from importlib.metadata import version
from packaging.version import Version
import warnings
import pandas as pd
import lightsim2grid

from lightsim2grid.gridmodel import init_from_pandapower
pp_ver_3 = Version(pandapower.__version__) >= Version("3")
tol = 1e-5

if pp_ver_3:
    from pandapower.pypower.idx_brch import F_BUS, T_BUS, BR_R, BR_X, BR_B, BR_G
else:
    from pandapower.pypower.idx_brch import F_BUS, T_BUS, BR_R, BR_X, BR_B
    
def _wye_delta_new(r, x, g, b, r_ratio=None, x_ratio=None):
    """taken from pandapower 3 source code"""
    if r_ratio is None:
        r_ratio = 0.5 + np.zeros_like(r)
    if x_ratio is None:
        x_ratio = 0.5 + np.zeros_like(x)
        
    tidx = (g != 0) | (b != 0)
    za_star = r[tidx] * r_ratio[tidx] + x[tidx] * x_ratio[tidx] * 1j
    zb_star = r[tidx] * (1 - r_ratio[tidx]) + x[tidx] * (1 - x_ratio[tidx]) * 1j
    zc_star = 1 / (g - 1j*b)[tidx]
    zSum_triangle = za_star * zb_star + za_star * zc_star + zb_star * zc_star
    zab_triangle = zSum_triangle / zc_star
    zac_triangle = zSum_triangle / zb_star
    zbc_triangle = zSum_triangle / za_star
    r[tidx] = zab_triangle.real
    x[tidx] = zab_triangle.imag
    yf = 1 / zac_triangle
    yt = 1 / zbc_triangle
    # 2 because in makeYbus Bcf, Bct are divided by 2:
    g[tidx] = yf.real * 2
    b[tidx] = yf.imag * 2
    g_asym = np.zeros_like(g)
    b_asym = np.zeros_like(b)
    g_asym[tidx] = 2 * yt.real - g[tidx]
    b_asym[tidx] = 2 * yt.imag - b[tidx]
    return r, x, g, b

print(f"pandapower version: {version('pandapower')}")
print(f"pandas version: {version('pandas')}")
print(f"numpy version: {version('numpy')}")
print(f"scipy version: {version('scipy')}")
print(f"lightsim2grid version: {version('lightsim2grid')}")

### main part of the code

# https://github.com/e2nIEE/pandapower/issues/2532
CASE = "case118"
net = pn.case118()
if pp_ver_3:
    net.trafo.i0_percent *= -1
    
# net.trafo.tap_pos = net.trafo.tap_neutral

pp.runpp(net, lightims2grid=False, numba=False)
with warnings.catch_warnings():
    warnings.filterwarnings("ignore")
    gridmodel = init_from_pandapower(net)
    
[... 137 lines cut ...]
print("Comparison ls2g ac_pf() and pandapower runpp")
V_ls = gridmodel.ac_pf(np.array([1.0]*len(net.bus)), 15, 1e-8)
Vm_ls = np.abs(V_ls)
Va_ls = np.angle(V_ls)

print(f"Vm max mismatch ls2g {lightsim2grid.__version__} vs. pandapower {pp.__version__}")
print((net.res_bus.vm_pu.values - Vm_ls.T).max())

print(f"Va max mismatch ls2g {lightsim2grid.__version__} vs. pandapower {pp.__version__}")
print((net.res_bus.va_degree.values - Va_ls.T).max())
# voltage angles are not matching between ls2g and pp?

branch_df_3 = pd.read_csv('ppc_branch_pp3.csv', index_col=0)
branch_df_2 = pd.read_csv('ppc_branch_pp2.csv', index_col=0)

BR_B3 = branch_df_3['BR_B'].values
BR_G3 = branch_df_3['BR_G'].values

BR_B2 = np.array(branch_df_2['BR_B'].values, dtype=np.complex128).real
BR_G2 = -np.array(branch_df_2['BR_B'].values.imag, dtype=np.complex128).imag

print(BR_B3[-len(net.trafo):]-BR_B2[-len(net.trafo):])
print(BR_G3[-len(net.trafo):]-BR_G2[-len(net.trafo):])

```

BDonnot, 2025-03-28:
> good idea, we can consider that when we re-work the json's

Thank you :-)

> The slight difference in voltage magnitude I would assume to be a numerical issue, but the difference in voltage angle between pp and ls2g is somewhat concerning. @BDonnot do you know why there might be this huge difference? Or do I compare apples with oranges here?

For the voltage magnitude I think you are absolutely right. This looks like numerical issues which we cannot control. Perfect :-)

For the voltage angle I would assume it's because either 1) the reference slack is difference or 2) the voltage angle at the slack is different (and it can be both 1 and 2)

> I will look deeper into the pandapower code to see where the differences come from. Will keep you posted.

Thank you very much :-) 

The difference might be between the T model and Pi model maybe ?

> What I don't really understand is why I get the same voltage magnitude and branch admittances/impedances after comparing ls2g with pp3.0 and pp2.4... Is there a dependency with pandapower when calling init_from_pandapower(net)? At first glance it looks like ls2g is calculating its branch values based on the pandas dataframes, not based on ppc in the aux_add_... functions, am I right?

If input coefficients are the same then lightsim2grid will lead to the same results. But if input coefficients (in the tables) are different then for sure lightsim2grid will lead to something different.

And you are correct ls2g has an internal modelling. Everything (for branch) is converted to a Pi equivalent model which is then used to computed the admittance matrix and compute the powerflow.

There is a function in lightsim2grid that can be use to compute the KCL mismatch (for each bus) if you want. Let me know if it can help you debug (it works by initializing a ls2g `gridmodel` and then passing a complex V voltage and ls2g returns the p and q mismatch at each buses)

mgovers, 2025-05-20:
the Power Grid Model also finds differences in the behavior relative to `pandapower==3.0.0`. Note that the behavior is the same between PGM and `pandapower<3`. See also https://github.com/PowerGridModel/power-grid-model-io/pull/303.

mgovers, 2025-06-02:
> the Power Grid Model also finds differences in the behavior (results) relative to `pandapower==3.0.0`. Note that the behavior is the same between PGM and `pandapower<3`. See also [PowerGridModel/power-grid-model-io#303](https://github.com/PowerGridModel/power-grid-model-io/pull/303).

seems like releasing v3.1 fixed the difference in behavior for power-grid-model. See also https://github.com/PowerGridModel/power-grid-model-io/pull/310