e2nIEE/pandapower #2564: [bug] JAO static grid model

gdigvijay, 2025-03-24:
### Bug report checklis

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [x] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp

net = pp.converter.from_jao(excel_file_path = r"202409_Core Static Grid Mode_6th release\model.xlsx", html_file_path = r"202409_Core Static Grid Mode_6th release\2024-09-13_Core_SGM_publication_files\2024-09-13_Core_SGM_publication.html")

print(net)
```

### Issue Description and Traceback

ValueError: For 3 Tranformers, no suitable bus location names were found, i.e. the algorithm did not find a (part) of Transformers-Location-Full Name that fits to Substation_1 or Substation_2 data in Lines or Tielines sheet.

### Expected Behavior

import pandapower as pp

net = pp.converter.from_jao(excel_file_path = r"202409_Core Static Grid Mode_6th release\model.xlsx", html_file_path = r"202409_Core Static Grid Mode_6th release\2024-09-13_Core_SGM_publication_files\2024-09-13_Core_SGM_publication.html")

print(net)

### Installed Versions

* python version: 3.10 * pandas version:  * networkx version:  * scipy version:  * numpy version:  * packaging version:  * tqdm version:  * deepdiff version:  * Operating System name/version:

### Label

- [x] Relevant labels are selected

vogt31337, 2025-03-28:
@gdigvijay can you provide the input data? Either as an attachment or as a link?

gdigvijay, 2025-03-28:
Here is the data I am using. Downloaded from [JAO](https://www.jao.eu/static-grid-model)

[202409_Core Static Grid Mode_6th release_gh.zip](https://github.com/user-attachments/files/19504120/202409_Core.Static.Grid.Mode_6th.release_gh.zip)

gdigvijay, 2025-04-02:
Fixed the issue. There is a typo in the dataset. Plus, the converter function cannot match upper case and lower case strings. Might be good to add that functionality. So TR != Tr.

closed 2025-04-02 by gdigvijay