e2nIEE/pandapower #2567: [bug] [plotting] [docs] pandapower.plotting.pf_res_plotly() cmap

gfritzler, 2025-03-27:
### Bug report checklis

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [x] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
import pandapower.plotting as plot

net = pp.networks.mv_oberrhein()

pp.runpp(net)

fig = plot.pf_res_plotly(net, cmap='algae',
                          line_width=3, bus_size=15,
                          climits_volt=(0.95, 1.05),
                          climits_load=(0, 150))

fig.show()
```

### Issue Description and Traceback

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Cell In[5], line 10
      6 pp.runpp(net)
      8 # https://matplotlib.org/stable/users/explain/colors/colormaps.html#sequential
---> 10 fig = plot.pf_res_plotly(net, cmap='algae',
     11                           line_width=3, bus_size=15,
     12                           climits_volt=(0.95, 1.05),
     13                           climits_load=(0, 150))
     15 fig.show()

File ~\PycharmProjects\a-project\.venv\Lib\site-packages\pandapower\plotting\plotly\pf_res_plotly.py:142, in pf_res_plotly(net, cmap, use_line_geo, on_map, projection, map_style, figsize, aspectratio, line_width, bus_size, climits_volt, climits_load, cpos_volt, cpos_load, filename, auto_open)
    136 hoverinfo = (
    137         net.line.name.astype(str) + '<br />' +
    138         'I = ' + net.res_line.loading_percent.round(precision).astype(str) + ' %' + '<br />' +
    139         'I_from = ' + net.res_line.i_from_ka.round(precision).astype(str) + ' kA' + '<br />' +
    140         'I_to = ' + net.res_line.i_to_ka.round(precision).astype(str) + ' kA' + '<br />').tolist()
    141 hoverinfo = pd.Series(index=net.line.index, data=hoverinfo)
--> 142 line_traces = create_line_trace(net, use_line_geo=use_line_geo, respect_switches=True,
    143                                 width=line_width,
    144                                 infofunc=hoverinfo,
    145                                 cmap=cmap_lines,
    146                                 cmap_vals=net.res_line['loading_percent'].values,
    147                                 cmin=climits_load[0],
    148                                 cmax=climits_load[1],
    149                                 cbar_title='Line Loading [%]',
    150                                 cpos=cpos_load)
    152 # ----- Trafos ------
    153 # hoverinfo which contains name and pf results
    154 hoverinfo = (
    155         net.trafo.name.astype(str) + '<br />' +
    156         'I = ' + net.res_trafo.loading_percent.round(precision).astype(str) + ' %' + '<br />' +
    157         'I_hv = ' + net.res_trafo.i_hv_ka.round(precision).astype(str) + ' kA' + '<br />' +
    158         'I_lv = ' + net.res_trafo.i_lv_ka.round(precision).astype(str) + ' kA' + '<br />').tolist()

File ~\PycharmProjects\a-project\.venv\Lib\site-packages\pandapower\plotting\plotly\traces.py:398, in create_line_trace(net, lines, use_line_geo, respect_switches, width, color, infofunc, trace_name, legendgroup, cmap, cbar_title, show_colorbar, cmap_vals, cmin, cmax, cpos, cmap_vals_category, hoverlabel, dash)
    395 node_element = "bus"
    396 separator_element = "switch"
[... 107 lines cut ...]
             'orrd', 'oryel', 'oxy', 'peach', 'phase', 'picnic', 'pinkyl',
             'piyg', 'plasma', 'plotly3', 'portland', 'prgn', 'pubu', 'pubugn',
             'puor', 'purd', 'purp', 'purples', 'purpor', 'rainbow', 'rdbu',
             'rdgy', 'rdpu', 'rdylbu', 'rdylgn', 'redor', 'reds', 'solar',
             'spectral', 'speed', 'sunset', 'sunsetdark', 'teal', 'tealgrn',
             'tealrose', 'tempo', 'temps', 'thermal', 'tropic', 'turbid',
             'turbo', 'twilight', 'viridis', 'ylgn', 'ylgnbu', 'ylorbr',
             'ylorrd'].
        Appending '_r' to a named colorscale reverses it.

### Installed Versions

* python version: 3.12.9
*  pandas version:  2.2.3
*  networkx version:  3.4.2
*  scipy version:  1.13.1
*  numpy version:  1.26.4
*  packaging version: 24.2
*  tqdm version: 6.4.2
* deepdiff version:  /
* Operating System name/version: Windows 11

### Label

- [ ] Relevant labels are selected

vogt31337, 2025-04-11:
@gfritzler, thanks for raising the issue. I think this is directly from plotly: https://plotly.com/python/builtin-colorscales/

Therefore I think we can't really repair that, since it seems to be a problem within plotly.

KS-HTK, 2025-04-11:
duplicate #2552

KS-HTK, 2025-04-11:
The issue is that there is some very unintuitive stuff going on here.

TL;DR: Markers are create using plotly colormaps, however line trace are created using matplotlib colormaps. So only colormaps in plotly and matplotlib can be used without issue.

`create_line_trace` calls `_create_branch_trace` which calls `get_plotly_color` and `get_plotly_cmap`. Both these functions use matplotlib and not plotly. This is very confusing as the function is called `get_plotly_cmap` but it tries to load a `mpl_colormap` and will throw a soft dependency error if matplotlib is not installed.