e2nIEE/pandapower #2581: [bug] AttributeError create_buses_dc parameter geodata

kornerc, 2025-04-08:
### Bug report checklis

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [x] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp

net = pp.create_empty_network()

pp.create_buses_dc(net=net, nr_buses_dc=2, vn_kv=30, geodata=[(1, 2), (3, 4)])
```

### Issue Description and Traceback

When using the parameter `geodata` of the function `create_buses_dc` an exception gets raised:
```
AttributeError: 'pandapowerNet' instance has no attribute 'bus_dc_geodata'
```

The same function for AC works:
```python
import pandapower as pp

net = pp.create_empty_network()

pp.create_buses(net=net, nr_buses=2, vn_kv=30, geodata=[(1, 2), (3, 4)])
```

### Expected Behavior

No exception gets raised

### Installed Versions

```
commit                : 0691c5cf90477d3503834d983f69350f250a6ff7
python                : 3.9.21
python-bits           : 64
OS                    : Windows
OS-release            : 10
Version               : 10.0.19045
machine               : AMD64
processor             : Intel64 Family 6 Model 142 Stepping 10, GenuineIntel
byteorder             : little
LC_ALL                : None
LANG                  : None
LOCALE                : English_United States.1252
pandas                : 2.2.3
numpy                 : 2.0.2
pytz                  : 2025.1
dateutil              : 2.9.0.post0
pip                   : 25.0.1
Cython                : None
sphinx                : 7.4.7
IPython               : 8.18.1
adbc-driver-postgresql: None
adbc-driver-sqlite    : None
bs4                   : None
blosc                 : None
bottleneck            : None
dataframe-api-compat  : None
fastparquet           : None
fsspec                : None
html5lib              : None
hypothesis            : None
gcsfs                 : None
jinja2                : 3.1.5
lxml.etree            : 5.3.1
matplotlib            : 3.9.4
numba                 : 0.60.0
numexpr               : None
odfpy                 : None
openpyxl              : 3.1.5
pandas_gbq            : None
psycopg2              : 2.9.10
pymysql               : None
pyarrow               : None
pyreadstat            : None
pytest                : 8.3.4
python-calamine       : None
pyxlsb                : None
s3fs                  : None
scipy                 : 1.13.1
sqlalchemy            : None
tables                : None
tabulate              : 0.9.0
xarray                : None
xlrd                  : None
xlsxwriter            : 3.2.2
zstandard             : None
tzdata                : 2025.1
qtpy                  : None
pyqt5                 : None
```

### Label

- [ ] Relevant labels are selected

vogt31337, 2025-04-11:
@kornerc, thanks for raising the issue! But we've switched from actual coordinates to geojson.

thomas-whaley-poco, 2025-04-28:
I assume this is caused in part by [commit 421b10d](https://github.com/e2nIEE/pandapower/commit/421b10d72a6cf986415fd87c9b94edc0790f26a8), where `"bus_geodata", "bus_dc_geodata", "line_geodata", "line_dc_geodata"` were removed from `create_empty_network(...)`.

`create_buses_dc` currently looks like:
```python
if geodata is not None:
    # works with a 2-tuple or a matching array
    net.bus_dc_geodata = pd.concat([
        net.bus_dc_geodata,
        pd.DataFrame(zeros((len(index), len(net.bus_dc_geodata.columns)), dtype=np.int64),
                        index=index, columns=net.bus_dc_geodata.columns)])
    net.bus_dc_geodata.loc[index, :] = nan
    net.bus_dc_geodata.loc[index, ["x", "y"]] = geodata
if coords is not None:
    net.bus_dc_geodata = pd.concat(
        [net.bus_dc_geodata, pd.DataFrame(index=index, columns=net.bus_dc_geodata.columns)])
    net["bus_dc_geodata"].loc[index, "coords"] = coords
```

This assumes that `net.bus_dc_geodata` exists in the first place, which it may not.