e2nIEE/pandapower #2595: Lack of clarity in the documentation on the mandatory ext_grid parameters for asymmetric 3-phase power flow.

RatonMagique, 2025-04-30:
In the documentation the variable `rx_max` is not flaged as mandatory for the 3 phases power flow calculations (c.f. screenshot). Nevertheless, when I'm not using the parameters in the power flow calculation, I get an error.  

**Document screenshot :** 
[image]

**Used script :**

        import pandapower as pp
        
        def test_run_3ph_power_flow():
            # Create an empty network suitable for 3-phase power flow
            net = pp.create_empty_network()
        
            # Create buses: HV (20 kV), LV (0.4 kV), and load bus (0.4 kV)
            bus1 = pp.create_bus(net, name="Bus HV", vn_kv=20.0, type="b")
            bus2 = pp.create_bus(net, name="Bus LV", vn_kv=0.4, type="b")
            bus3 = pp.create_bus(net, name="Bus Load", vn_kv=0.4, type="b")
        
            # Connect external grid with short-circuit parameters (balanced source)
            pp.create_ext_grid(
                net, bus=bus1, vm_pu=1.0, name="Grid Connection",
                s_sc_max_mva=500,
                # rx_max=0.1, # Disabled parameters
                x0x_max=3.0, r0x0_max=0.5
            )
        
            # Create a three-phase transformer with zero-sequence parameters
            pp.create_transformer_from_parameters(
                net, hv_bus=bus1, lv_bus=bus2,
                sn_mva=0.4, vn_hv_kv=20.0, vn_lv_kv=0.4,
                vk_percent=6, vkr_percent=0.7, pfe_kw=0.4, i0_percent=0.1, 
                shift_degree=-330, vector_group='Dyn11',
                tap_side="hv", tap_neutral=0, tap_min=-2, tap_max=2,
                tap_step_percent=2.5, tap_pos=0, name="Trafo",
                vk0_percent=4.0, vkr0_percent=0.5,
                mag0_percent=0.1, mag0_rx=10.0,
                si0_hv_partial=1
            )
        
            # Create a line with zero- and positive-sequence parameters
            pp.create_line_from_parameters(
                net, from_bus=bus2, to_bus=bus3, length_km=0.05,
                r_ohm_per_km=0.642, x_ohm_per_km=0.083, c_nf_per_km=210,
                r0_ohm_per_km=0.9, x0_ohm_per_km=0.35, c0_nf_per_km=105,
                max_i_ka=0.14, name="Line"
            )
        
            # Add an unbalanced 3-phase load
            pp.create_asymmetric_load(
                net, bus=bus3,
                p_a_mw=0.015, q_a_mvar=0.005,
                p_b_mw=0.020, q_b_mvar=0.007,
                p_c_mw=0.010, q_c_mvar=0.003,
                name="Unbalanced Load"
            )
        
            # Add an unbalanced PV generator
            pp.create_asymmetric_sgen(
                net, bus=bus3,
                p_a_mw=0.005, q_a_mvar=0.0,
                p_b_mw=0.005, q_b_mvar=0.0,
                p_c_mw=0.005, q_c_mvar=0.0,
                name="Unbalanced PV"
            )
        
            # Run 3-phase power flow
            pp.runpp_3ph(net)
        
            # Print results
            print(net.res_bus_3ph)
            print(net.res_asymmetric_load)
            print(net.res_asymmetric_sgen)
 

**Error:**

        def _add_ext_grid_sc_impedance(net, ppc):
                mode = net._options["mode"]
                bus_lookup = net["_pd2ppc_lookups"]["bus"]
                if mode == "sc":
                    case = net._options["case"]
                else:
                    case = "max"
                eg = net["ext_grid"][net._is_elements["ext_grid"]]
                if len(eg) == 0:
                    return
                eg_buses = eg.bus.values
                eg_buses_ppc = bus_lookup[eg_buses]
            
                if mode == "sc":
                    c = ppc["bus"][eg_buses_ppc, C_MAX] if case == "max" else ppc["bus"][eg_buses_ppc, C_MIN]
                else:
                    c = 1.1
                if not "s_sc_%s_mva" % case in eg:
                    raise ValueError(("short circuit apparent power s_sc_%s_mva needs to be specified for "
                                      "external grid \n Try: net.ext_grid['s_sc_max_mva'] = 1000") % case)
                s_sc = eg["s_sc_%s_mva" % case].values/ppc['baseMVA']
                if not "rx_%s" % case in eg:
        >           raise ValueError(("short circuit R/X rate rx_%s needs to be specified for external grid \n"
                                      " Try: net.ext_grid['rx_max'] = 0.1") % case)
        E           ValueError: short circuit R/X rate rx_max needs to be specified for external grid 
        E            Try: net.ext_grid['rx_max'] = 0.1
`

pawellytaev, 2025-05-08:
Hi @RatonMagique , thanks for raising this. Do you see this also for other elements in asymmetric power flow? In case of the `ext_grid` the `rx_max` should be labeled with four stars, i.e., `rx_max****`

pawellytaev, 2025-05-08:
P.S. same holds for `rx_min`