e2nIEE/pandapower #2598: [feature] Creating a transformer from a standard type should automatically apply the vector group.

RatonMagique, 2025-05-07:
### Feature Checklist

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [x] Adding new functionality to pandapower

- [ ] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

While creating a transformer standard type, we can specify the vector group, which provides information about the winding configuration on the HV and LV sides. This information is also required to run a three-phase power flow analysis.
However, after creating a new transformer from a standard type, it is necessary to re-enter the vector group manually, as it is not automatically populated from the standard type.

### Feature Description

If the vector group information were included in the standard type, it would be more convenient to have it automatically filled when creating a transformer.

### Label

- [ ] Relevant labels are selected

pawellytaev, 2025-05-08:
Hey @RatonMagique, thanks for opening this issue! Could you provide a minimal example code for this issue?

RatonMagique, 2025-05-09:
Hi, yes here is a code example : 

While running this simple piece of code, I use the vector group "Dyn" for my tranformer standard type. However, this parameter is absent from the trafo dataframe (c.f. csv file) and the following error is raised while running. 

**Code :** 
```
  import pandapower as pp
  
  def test_run_3ph_power_flow():
      # Create an empty network suitable for 3-phase power flow
      net = pp.create_empty_network()
  
      # Create buses: HV (20 kV), LV (0.4 kV), and load bus (0.4 kV)
      bus1 = pp.create_bus(net, name="Bus HV", vn_kv=20.0, type="b")
      bus2 = pp.create_bus(net, name="Bus LV", vn_kv=0.4, type="b")
      bus3 = pp.create_bus(net, name="Bus Load", vn_kv=0.4, type="b")
  
      # Connect external grid with short-circuit parameters (balanced source)
      pp.create_ext_grid(
          net, bus=bus1, vm_pu=1.0, name="Grid Connection",
          s_sc_max_mva=500,
          rx_max=0.1,
          x0x_max=3.0, r0x0_max=0.5
      )
  
      # Create transformer's standard type
      pp.create_std_type(net, {
          "sn_mva": 0.4,
          "vn_hv_kv": 20.0,
          "vn_lv_kv": 0.4,
          "vk_percent": 6,
          "vkr_percent": 0.7,
          "pfe_kw": 0.4,
          "i0_percent": 0.1,
          "shift_degree" : 30,
          "vector_group" : "Dyn", # The vector group is added here
          "vk0_percent": 4.0,
          "vkr0_percent": 0.5,
          "mag0_percent": 0.1,
          "mag0_rx": 10.0,
          "si0_hv_partial": 1.0
      }, name="test_transfo_type", element="trafo")
  
  
      pp.create_transformer(
          net, bus1, bus2, name="Transform 1", std_type="test_transfo_type",)
  
      # Create a line with zero- and positive-sequence parameters
      pp.create_line_from_parameters(
          net, from_bus=bus2, to_bus=bus3, length_km=0.05,
          r_ohm_per_km=0.642, x_ohm_per_km=0.083, c_nf_per_km=210,
          r0_ohm_per_km=0.9, x0_ohm_per_km=0.35, c0_nf_per_km=105,
          max_i_ka=0.14, name="Line"
      )
  
      # Add an unbalanced 3-phase load
      pp.create_asymmetric_load(
          net, bus=bus3,
          p_a_mw=0.015, q_a_mvar=0.005,
          p_b_mw=0.020, q_b_mvar=0.007,
          p_c_mw=0.010, q_c_mvar=0.003,
          name="Unbalanced Load"
      )
  
      # Add an unbalanced PV generator
      pp.create_asymmetric_sgen(
          net, bus=bus3,
          p_a_mw=0.005, q_a_mvar=0.0,
          p_b_mw=0.005, q_b_mvar=0.0,
          p_c_mw=0.005, q_c_mvar=0.0,
          name="Unbalanced PV"
      )
  
      # Run 3-phase power flow
      pp.runpp_3ph(net)
  
      # Print results
      print(net.res_bus_3ph)
      print(net.res_asymmetric_load)
      print(net.res_asymmetric_sgen)
```

**Error :**

```
          if "vector_group" not in trafo_df:
>           raise ValueError("Vector Group of transformer needs to be specified for zero "
                             "sequence modelling \n Try : net.trafo[\"vector_group\"] = 'Dyn'")
E           ValueError: Vector Group of transformer needs to be specified for zero sequence modelling 
E            Try : net.trafo["vector_group"] = 'Dyn'
```

[net_trafo.csv](https://github.com/user-attachments/files/20116502/net_trafo.csv)

pawellytaev, 2025-05-09:
thanks for the code snippet. it is fixed in #2603 by adding vector_group to the zero sequence parameters in create_transformer()

PR #2603 (2025-06-16): Fix/#2598 vector groups in standard type
fixes #2598
closed 2025-05-09 by pawellytaev