e2nIEE/pandapower #2611: Pandapower <-> NEPLAN

the-vividlemon, 2025-05-15:
### Feature Checklist

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hello,

I work for a Swiss DSO and use pandapower extensively to model our LV network. We rely on pandapower to build our digital twin and assess various consumption and generation scenarios.

In some cases, we would like to test alternative LV topologies—for example, switching from a long radial distribution line to a point-to-point configuration. From my experience with pandapower, implementing such topological changes directly seems complex or not straightforward.

One idea we are considering is to use NEPLAN to make the topological modifications, and then import the modified network back into pandapower for further analysis.

Do you have any suggestions on how best to transfer network data between NEPLAN and pandapower? Or alternatively, is there a recommended way to implement such topology changes directly within pandapower?

Thank you in advance for your support and for this great tool!

### Label

- [ ] Relevant labels are selected

the-vividlemon, 2025-05-16:
Hello Alex,
Thank you for your reply. I think a quick call will be the easiest https://teams.live.com/l/invite/FEA4A6Wgt3UGEcxEwE :)

heckstrahler, 2025-11-27:
Closing for now, if there is still interest please write to the contact information at https://www.iee.fraunhofer.de/en/research-fields/grid-planning-grid-operation.html