e2nIEE/pandapower #2624: [bug] IndexError when executing power flow with distributed_slack=True with xward elements in net

panos-xenos, 2025-05-29:
### Bug report checklis

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [x] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp
if not net.xward.empty:
    pp.runpp(net, distributed_slack=True)
```

### Issue Description and Traceback

Getting IndexError when trying to execute a power flow with distributed_slack=True for a network case that contains xward elements. Here's the output message:
```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[95], line 1
----> 1 pp.runpp(net, distributed_slack=True)

File D:\repos_personal\pandapower-develop\pandapower\pandapower\run.py:252, in runpp(net, algorithm, calculate_voltage_angles, init, max_iteration, tolerance_mva, trafo_model, trafo_loading, enforce_q_lims, check_connectivity, voltage_depend_loads, consider_line_temperature, run_control, distributed_slack, tdpf, tdpf_delay_s, **kwargs)
    250 _check_bus_index_and_print_warning_if_high(net)
    251 _check_gen_index_and_print_warning_if_high(net)
--> 252 _powerflow(net, **kwargs)

File D:\repos_personal\pandapower-develop\pandapower\pandapower\powerflow.py:62, in _powerflow(net, **kwargs)
     57 net._pd2ppc_lookups = {"bus": array([], dtype=int64), "bus_dc": array([], dtype=int64),
     58                        "ext_grid": array([], dtype=int64), "gen": array([], dtype=int64),
     59                        "branch": array([], dtype=int64), "branch_dc": array([], dtype=int64)}
     61 # convert pandapower net to ppc
---> 62 ppc, ppci = _pd2ppc(net, **kwargs)
     64 # store variables
     65 net["_ppc"] = ppc

File D:\repos_personal\pandapower-develop\pandapower\pandapower\pd2ppc.py:208, in _pd2ppc(net, sequence, **kwargs)
    205 if "pf" in mode or "se" in mode:
    206     _check_for_reference_bus(ppc)
--> 208 _build_gen_ppc(net, ppc)
    210 _replace_nans_with_default_limits(net, ppc)
    212 # generates "internal" ppci format (for powerflow calc)
    213 # from "external" ppc format and updates the bus lookup
    214 # Note: Also reorders buses and gens in ppc

File D:\repos_personal\pandapower-develop\pandapower\pandapower\build_gen.py:67, in _build_gen_ppc(net, ppc)
     65 xward_pq_buses = _get_xward_pq_buses(net, ppc)
     66 gen_mask, xward_mask = _gen_xward_mask(net, ppc)
---> 67 _normalise_slack_weights(ppc, gen_mask, xward_mask, xward_pq_buses)

File D:\repos_personal\pandapower-develop\pandapower\pandapower\build_gen.py:446, in _normalise_slack_weights(ppc, gen_mask, xward_mask, xward_pq_buses)
    444 for subnet in subnets:
    445     subnet_gen_mask = np.isin(gen_buses, subnet)
--> 446     sum_slack_weights = np.sum(slack_weights_gen[subnet_gen_mask])
    447     if np.isclose(sum_slack_weights, 0):
    448         # ppc['gen'][subnet_gen_mask, SL_FAC] = 0
    449         raise ValueError("Distributed slack contribution factors in "
    450                          "island '%s' sum to zero." % str(subnet))

IndexError: boolean index did not match indexed array along axis 0; size of axis is 232 but size of corresponding boolean axis is 377
```

The issue is within `build_gen.py` after the conversion into a ppc structure that represents an xward element with a 'gen'  and a 'branch' row. The core of the mismatch is here:
```
gen_buses = ppc['gen'][gen_mask, GEN_BUS].astype(np.int64)
# …
gen_buses = np.r_[gen_buses, xward_pq_buses]
slack_weights_gen = np.r_[ppc['gen'][gen_mask, SL_FAC], ppc['gen'][xward_mask, SL_FAC]].astype(np.float64)
```

`gen_buses` is built by concatenating the buses of the filtered gens (`gen_mask`) plus the PQ buses of the xwards (`xward_pq_buses`). `slack_weights_gen` is built by concatenating the weights of the filtered gens (`gen_mask`) plus the weights of the xward gens (`xward_mask`). If `len(xward_pq_buses) != np.count_nonzero(xward_mask)`, the two arrays will have different lengths, raising an IndexError. The difference I'm getting of 377-232=145 is actually the number of xwards in my network case, so it seems these are double-counted somehow.

### Expected Behavior

Power flow should run successfully with distributed_slack=True for a case that contains xward elements.

### Installed Versions

INSTALLED VERSIONS
------------------
commit                : 0691c5cf90477d3503834d983f69350f250a6ff7
python                : 3.11.7
python-bits           : 64
OS                    : Windows
OS-release            : 10
Version               : 10.0.26100
machine               : AMD64
processor             : AMD64 Family 25 Model 116 Stepping 1, AuthenticAMD
byteorder             : little
LC_ALL                : None
LANG                  : None
LOCALE                : English_United States.1252
pandas                : 2.2.3
numpy                 : 2.0.2
pytz                  : 2024.2
dateutil              : 2.9.0.post0
pip                   : 24.3.1
Cython                : None
sphinx                : 8.1.3
IPython               : 8.30.0
adbc-driver-postgresql: None
adbc-driver-sqlite    : None
bs4                   : 4.12.3
blosc                 : None
bottleneck            : None
dataframe-api-compat  : None
fastparquet           : None
fsspec                : None
html5lib              : None
hypothesis            : None
gcsfs                 : None
jinja2                : 3.1.5
lxml.etree            : 5.3.0
matplotlib            : 3.9.3
numba                 : 0.60.0
numexpr               : None
odfpy                 : None
openpyxl              : 3.1.5
pandas_gbq            : None
psycopg2              : 2.9.10
pymysql               : None
pyarrow               : None
pyreadstat            : None
pytest                : 8.3.4
python-calamine       : None
pyxlsb                : None
s3fs                  : None
scipy                 : 1.13.1
sqlalchemy            : None
tables                : None
tabulate              : 0.9.0
xarray                : None
xlrd                  : None
xlsxwriter            : None
zstandard             : None
tzdata                : 2024.2
qtpy                  : None
pyqt5                 : None

### Label

- [x] Relevant labels are selected

vogt31337, 2025-07-23:
@panos-xenos, could you provide a better example?

panos-xenos, 2025-07-30:
@vogt31337 I tried to create a small test case but I failed to get the same error. Due to confidentiality, we can send you a full network case that presents the issue via email through secure fileshare. Is that ok?

vogt31337, 2025-11-25:
@panos-xenos , I finally found the Issue. But I cannot estimate if it is now correct. Maybe we should discuss this in a call.
My solution:

```Python
net.xward.slack_weight = 0.
net.gen.slack_weight = net.gen.slack_weight.fillna(0.)

if not net.xward.empty:
    pp.runpp(net, distributed_slack=True)

```

So from what I understood, xward wasn't used as a slack, but all the values were np.nan. Which I've just replaced with 0. and the more pressing issue, in build_gen.py line 458ff:
```
# raise NotImplementedError if there are several separate zones for distributed slack:
if not np.isclose(sum(ppc['bus'][:, SL_FAC_BUS]), 1):
    raise NotImplementedError("Distributed slack calculation is not implemented for several separate zones at once,"
                              "please calculate the zones separately.")

```
Which basically sums the "zones" with slack weights attached to it. That's why in line 441 same file, the system iterates over the subnets pp found.

**Another important 'fix'**, line 430 `gen_buses = np.r_[gen_buses, xward_pq_buses]` must be commented out, since otherwise gen_buses is too large and indexing will fail. I think rbolgaryn thought these two lists need to be stitched together, but xward_pq_buses are already in the ppc['gen'] table.

I am not sure if this is just a fix for your Issue or a general one...

panos-xenos, 2025-11-26:
Hey @vogt31337, many thanks for investigating and getting back to me! This seems to do the trick, yes. Do you think the documentation should reflect that somewhere?

vogt31337, 2026-11-12:
Tried this again:
```python
import pandapower as pp
net = pp.pandapowerNet(name="test")
if not net.xward.empty:
    pp.runpp(net, distributed_slack=True)
```

runs without any problems, issue is fixed.

closed 2026-11-12 by vogt31337