e2nIEE/pandapower #2627: Modelating a battery (BESS)

joseptrilla, 2025-05-30:
### Feature Checklist

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [ ] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

### Issue

Hi,
I have successfully modeled an electrical network using pandapower, and the simulation runs smoothly. However, I’m currently facing difficulties with modeling a Battery Energy Storage System (BESS). I’ve attempted to model it using the ‘storage’ element and implemented a script that runs an Optimal Power Flow (OPF) analysis 24 times, updating the state of charge in each iteration.

Despite this setup, the simulation doesn’t seem to acknowledge the presence of the BESS, and I’m unsure what I might be missing or doing incorrectly.

### Label

- [ ] Relevant labels are selected

pawellytaev, 2025-07-15:
Hi @joseptrilla, it is hard to judge what could be wrong without having an example code, could you provide a small example of your problem? :-)

have you also checked out the OPF section on storages in the documentation, e.g., have you used the function pandapower.opf_task()? see here: [https://pandapower.readthedocs.io/en/latest/opf/pypower_run.html#optimization-with-pypower](https://pandapower.readthedocs.io/en/latest/opf/pypower_run.html#optimization-with-pypower)

as an alternative to using the storage element, you could try using the sgen or load element, connect them to the same bus and activate or deactivate them, depending on charging or discharging the BESS?

vogt31337, 2025-12-02:
@joseptrilla , we have a storage optimization module using pandamodels / powermodels. There is also an example in a jupyter notebook in the tutorials. I just fixed the stuff, and it is usable again, when the PR #2749 is merged.

vogt31337, 2025-12-03:
#2749 was merged.