e2nIEE/pandapower #2662: [bug] Inconsistent currents and powers on the LV side of transformer in 3 phase power flow

YassineAbdelouadoud, 2025-07-11:
### Bug report checklis

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar reports

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for usefull code snippets and examples of use

- [x] Reproduced the issue after updating with `pip install --upgrade pandapower` (or `git pull`)

- [x] Tried basic troubleshooting (if a bug/error) like restarting the interpreter and checking the pythonpath

### Reproducible Example

```python
import pandapower as pp

# Create network
net = pp.create_empty_network()

# Add buses
bus_hv = pp.create_bus(net, vn_kv=20, name="20kV slack bus")
bus_lv_0 = pp.create_bus(net, vn_kv=0.4, name="400V trafo LV side bus")
bus_lv_1 = pp.create_bus(net, vn_kv=0.4, name="400V load bus")

# Add external grid (slack bus) to HV bus
pp.create_ext_grid(
    net,
    bus=bus_hv,
    vm_pu=1.0,
    va_degree=0,
    s_sc_max_mva=1000,
    rx_max=0.1,
    x0x_max=3.0,
    r0x0_max=0.5,
)

# Create Dyn transformer
pp.create_transformer_from_parameters(
    net,
    hv_bus=bus_hv,
    lv_bus=bus_lv_0,
    sn_mva=0.25,
    vn_hv_kv=20,
    vn_lv_kv=0.4,
    vk_percent=5,
    vk0_percent=5.,
    vkr_percent=2.,
    vkr0_percent=2.,
    mag0_percent=10.,
    mag0_rx=1.,
    pfe_kw=5.,          # Set iron losses to ZERO
    i0_percent=2.,       # Set no-load current to ZERO
    vector_group="Dyn",
    shift_degree=330.,
    si0_hv_partial=0.5
)

# Create LV line
pp.create_line_from_parameters(
    net,
    from_bus=bus_lv_0,
    to_bus=bus_lv_1,
    length_km=0.1,
    r0_ohm_per_km=0.789,
    x0_ohm_per_km=0.306,
    c0_nf_per_km=272.9,
    max_i_ka=0.496,
    r_ohm_per_km=0.184,
    x_ohm_per_km=0.1900664,
    c_nf_per_km=273,
)

# Create unbalanced load (only phase A)
pp.create_asymmetric_load(
    net,
    bus=bus_lv_1,
    p_a_mw=0.2,   # Active power on phase A
    q_a_mvar=0.,
    p_b_mw=0.,     # Zero load on phase B
    q_b_mvar=0.,
    p_c_mw=0.,     # Zero load on phase C
    q_c_mvar=0.
)

# Run THREE-PHASE power flow
pp.runpp_3ph(net, calculate_voltage_angles=True)

# Check LV side currents
print("trafo LV currents [kA]:")
print(net.res_trafo_3ph[['i_a_lv_ka', 'i_b_lv_ka', 'i_c_lv_ka']])

# Check LV side active power
print("\ntrafo LV active Power [MW]:")
print(net.res_trafo_3ph[['p_a_lv_mw', 'p_b_lv_mw', 'p_c_lv_mw']])

# Check LV side currents
print("\nline from currents [kA]:")
print(net.res_line_3ph[['i_a_from_ka', 'i_b_from_ka', 'i_b_from_ka']])

# Check LV side active power
print("\nline from active Power [MW]:")
print(net.res_line_3ph[['p_a_from_mw', 'p_b_from_mw', 'p_c_from_mw']])
```

### Issue Description and Traceback

The results of the reproducible example are the following : 

trafo LV currents [kA]:
   i_a_lv_ka  i_b_lv_ka  i_c_lv_ka
0   1.296891   0.000002   0.000002

trafo LV active Power [MW]:
   p_a_lv_mw  p_b_lv_mw  p_c_lv_mw
0  -0.176577  -0.022445  -0.072996

line from currents [kA]:
   i_a_from_ka  i_b_from_ka  i_b_from_ka
0     1.296891     0.000002     0.000002

line from active Power [MW]:
   p_a_from_mw   p_b_from_mw   p_c_from_mw
0     0.264866  2.722787e-08 -1.814035e-08

Notice the inconsistencies in the currents and powers on the LV side of the transformer while the power and currents are consistent at the from side of the LV line.

I have found this issue : 
[issue 1021](https://github.com/e2nIEE/pandapower/issues/1021)
and the following fix : 
[pull 1343](https://github.com/e2nIEE/pandapower/pull/1343)

Is it possible that pull #1343 fix the issue for currents but not powers ? 

### Expected Behavior

In this case, powers at the LV side of the trafo should match powers at the from side of the line (and also be equal to current * voltage)

### Installed Versions

pandapower version is 3.1.2

output of pd.show_versions() : 

INSTALLED VERSIONS
------------------
commit                : 0691c5cf90477d3503834d983f69350f250a6ff7
python                : 3.12.9
python-bits           : 64
OS                    : Linux
OS-release            : 5.15.0-139-generic
Version               : #149-Ubuntu SMP Fri Apr 11 22:06:13 UTC 2025
machine               : x86_64
processor             : x86_64
byteorder             : little
LC_ALL                : None
LANG                  : fr_FR.UTF-8
LOCALE                : fr_FR.UTF-8
pandas                : 2.2.3
numpy                 : 1.26.4
pytz                  : 2024.1
dateutil              : 2.9.0.post0
pip                   : 25.0.1
Cython                : None
sphinx                : None
IPython               : None
adbc-driver-postgresql: None
adbc-driver-sqlite    : None
bs4                   : None
blosc                 : None
bottleneck            : None
dataframe-api-compat  : None
fastparquet           : None
fsspec                : None
html5lib              : None
hypothesis            : None
gcsfs                 : None
jinja2                : None
lxml.etree            : 5.3.1
matplotlib            : 3.10.1
numba                 : 0.61.2
numexpr               : None
odfpy                 : None
openpyxl              : None
pandas_gbq            : None
psycopg2              : None
pymysql               : None
pyarrow               : None
pyreadstat            : None
pytest                : 8.3.5
python-calamine       : None
pyxlsb                : None
s3fs                  : None
scipy                 : 1.13.1
sqlalchemy            : None
tables                : None
tabulate              : None
xarray                : None
xlrd                  : None
xlsxwriter            : 3.2.2
zstandard             : None
tzdata                : 2025.1
qtpy                  : None
pyqt5                 : None

### Label

- [x] Relevant labels are selected

furqan463, 2025-08-04:
Pull#1343 solved the issue of inconsistency in TF LV and LV_Line currents but in turn created an inconsistency in TF current vs Power.
As Explained in issue#1021, in case of Dy transformer, delta-wye conversions are being applied and both HV and LV results are w.r.t HT side. If LV power is to be made consistent with LV_Line power, there'll be an inconsistency between TF HT and LV Powers. This inconsistency is present inherently in some vector groups. Zero Sequence on LV side do not appear on HV side of the Dyn Transformer.

furqan463, 2025-10-17:
@KS-HTK @vogt31337 please close this issue as it is resolved in PR #2720