e2nIEE/pandapower #2715: [feature] Parallel Contigency analysis

fbonhoff, 2025-09-18:
### Feature Checklist

- [x] Searched the [issues page](https://github.com/e2nIEE/pandapower/issues) for similar feature requests

- [x] Read the relevant sections of the [documentation](https://pandapower.readthedocs.io/en/latest/about.html)

- [x] Browse the repository, [tutorials](https://github.com/e2nIEE/pandapower/tree/develop/tutorials) and [tests](https://github.com/e2nIEE/pandapower/tree/develop/pandapower/test) for already existing functionalities

### Feature Type

- [ ] Adding new functionality to pandapower

- [x] Changing existing functionality in pandapower

- [ ] Removing existing functionality in pandapower

### Problem Description

I was doing some analysis on a network using contingency analysis and I think using multiprocessing can be a quick fix to parallelize the calculation. 

As an example I want to do an optimal placement or configuration of a problem using an optimization algorithm which needs many full contingency analysis taking hours to calculate because its running only on a single core.

P.s. This is my first issue, so if I am doing something wrongly please leave feedback on how i can improve contributing.

### Feature Description

Quickly described every process just runs its own Powerflow and updates the result network. 
I did a quick implementation, will probably push it to a Fork later today, just wanted to add the feature request before hand. 

Intial Test runs on a 2500 bus network gave me those results, showing pretty good scaling:

16 cores
133.15 Sekunden
8 cores
242.65 Sekunden
4 cores
479.52 Sekunden
2 cores:
944.85 Sekunden
1 Core
1955.82 Sekunden

All that changes is an additional parameter in the contingency function :

Before: 
res = pp.contingency.run_contingency(net, nminus1_cases, pf_options=options,pf_options_nminus1=options)

After:
res = run_contingency_parallel(net, nminus1_cases, pf_options=options,pf_options_nminus1=options, n_procs=4)

So when not set nothing changes, leading to an in-place replacement.

### Label

- [ ] Relevant labels are selected

fbonhoff, 2025-09-18:
Just pushed my version onto an fork. 
I tested the analysis on the 2869pegase network with theses results (I attached my "test script" as well. 

Serial time: 258.5909569263458
---------------------------------------------------
Parallel time with 1 process: 265.6016902923584
---------------------------------------------------
Parallel time with 2 processes: 143.68792581558228
---------------------------------------------------
Parallel time with 4 processes: 70.4729425907135
---------------------------------------------------
Parallel time with 8 processes: 36.287999391555786
---------------------------------------------------
Parallel time with 16 processes: 19.254905223846436

There is a slight overhead but, the improvement is significant in my opinion. 

[contingency.py](https://github.com/user-attachments/files/22408503/contingency.py)

vogt31337, 2025-09-26:
tbh, @fbonhoff this looks really nice. I've just checked your development branch. Probably there is the possibility to speed it up even further. Maybe by using a clever topology analysis or something similar.

fbonhoff, 2025-09-27:
Mhh i will create a pull request for the contingency. 

Not sure what you mean with topology analysis. Filtering out "critical lines" in case of like lines that connect to independent circles, or something like that? 

I would consider that an optional pre processing step, as it is highly network dependent and task dependent if it matters and if a speed up is realistic.

fbonhoff, 2026-03-09:
We may need to reopen this… I found a bug in the way the network is accessed when using especially large scale parallelism which interferes with the cases being distinct and causes false line failures at some point. I think I have a fix, will give an update as well, later this week. With a detailed problem description.

closed 2026-11-09 by vogt31337